nfcpy/nfcpy #32: Communication fails with PN532 on UART

nehpetsde, 2016-04-26:
I have tried to read a NFC tag with my PN532 plugged the UART of my raspberry pi.

It works fine with nfc-poll from the nfclib.
Nevertheless, with nfcpy, cloned today (14 august), it crashes with the following error message :

pi@raspberrypi ~/nfcpy $ python examples/helloworld.py

INFO:nfc.clf:searching for reader with path 'tty:AMA0:pn53x'
INFO:nfc.clf:using NXP PN532 at /dev/ttyAMA0
Please touch a tag to send a hello to the world
WARNING:nfc.dev:Driver.set_communication_mode() should be implemented.
WARNING:nfc.dev.pn53x:[PN53x Error 0x13] Format error during RF communication

Reproduction steps :
 * with PN532 on UART
 * change nfc.ContactlessFrontend('tty:AMA0:pn53x') in hellopword.py
 * launch the helloword script
 * scan a tag

nehpetsde, 2016-04-26:
(by stephen-tiedemann)
I need more information: Please try to read the tag with "python examples/tagtool.py show --device tty:AMA0:pn53x -d nfc.dev" and post the debug output here.

nehpetsde, 2016-04-26:
(by matthieu-falce)
I don't know why, but the line you gave me doesn't work. I can't put anything after show. 
Even 'python examples/tagtool.py show -h' has only 2 options while 'python examples/tagtool.py -h' shows the complete command line option... 

A "funny" thing that happens is that I have a LED on the RFID device that is lit on during the initialization and the release of the device.  When I launch the script, the LED doesn't stop flickering very quickly. 

Here is the log you wanted : 
pi@raspberrypi ~/nfcpy $ python examples/tagtool.py --device tty:AMA0:pn53x -d nfc.dev show
[main] enable debug output for module 'nfc.dev'
[nfc.clf] searching for reader with path 'tty:AMA0:pn53x'
[nfc.dev] trying pn53x on 'tty:AMA0:pn53x'
[nfc.dev] import nfc.dev.pn53x
[nfc.dev.transport] >>> 0000ff00ff00
[nfc.dev.pn53x] GetFirmwareVersion command with timeout 100 ms
[nfc.dev.transport] >>> 0000ff02fed4022a00
[nfc.dev.transport] <<< 0000ff06fad50332010607e800
[nfc.dev.pn53x] chipset is a PN532 version 1.6
[nfc.dev.pn53x] Diagnose command with timeout 100 ms
[nfc.dev.transport] >>> 0000ff08f8d400006e666370790c00
[nfc.dev.transport] <<< 0000ff08f8d501006e666370790a00
[nfc.dev.pn53x] ATR_RES timeout:   102.4 ms
[nfc.dev.pn53x] non-DEP timeout:    51.2 ms
[nfc.dev.pn53x] RFConfiguration command with timeout 100 ms
[nfc.dev.transport] >>> 0000ff06fad432020b0b0ad800
[nfc.dev.transport] <<< 0000ff02fed533f800
[nfc.dev.pn53x] set retries: ATR_REQ=2 PSL_REQ=1 PassiveTarget=3
[nfc.dev.pn53x] RFConfiguration command with timeout 100 ms
[nfc.dev.transport] >>> 0000ff06fad43205020103ef00
[nfc.dev.transport] <<< 0000ff02fed533f800
[nfc.clf] using NXP PN532 at /dev/ttyAMA0
[main] touch a tag
[nfc.dev.pn53x] polling for NFC-A technology
[nfc.dev.pn53x] InListPassiveTarget command with timeout 1000 ms
[nfc.dev.transport] >>> 0000ff04fcd44a0100e100
[nfc.dev.transport] <<< 0000ff03fdd54b00e000
[nfc.dev.pn53x] InListPassiveTarget command with timeout 1000 ms
[nfc.dev.transport] >>> 0000ff04fcd44a0104dd00
[nfc.dev.transport] <<< 0000ff03fdd54b00e000
[nfc.dev.pn53x] poll NFC-F 00ffff0103
[nfc.dev.pn53x] InListPassiveTarget command with timeout 1000 ms
[nfc.dev.transport] >>> 0000ff09f7d44a010200ffff0103dd00
[nfc.dev.transport] <<< 0000ff03fdd54b00e000
[nfc.dev.pn53x] poll NFC-F 00ffff0103
[nfc.dev.pn53x] InListPassiveTarget command with timeout 1000 ms
[nfc.dev.transport] >>> 0000ff09f7d44a010100ffff0103de00
[nfc.dev.transport] <<< 0000ff03fdd54b00e000
[nfc.dev.pn53x] RFConfiguration command with timeout 100 ms
[nfc.dev.transport] >>> 0000ff04fcd4320100f900
[nfc.dev.transport] <<< 0000ff02fed533f800
[nfc.dev.pn53x] polling for NFC-A technology
[nfc.dev.pn53x] InListPassiveTarget command with timeout 1000 ms
[nfc.dev.transport] >>> 0000ff04fcd44a0100e100
[nfc.dev.transport] <<< 0000ff03fdd54b00e000
[nfc.dev.pn53x] InListPassiveTarget command with timeout 1000 ms
[nfc.dev.transport] >>> 0000ff04fcd44a0104dd00
[nfc.dev.transport] <<< 0000ff03fdd54b00e000
[nfc.dev.pn53x] poll NFC-F 00ffff0103
[nfc.dev.pn53x] InListPassiveTarget command with timeout 1000 ms
[nfc.dev.transport] >>> 0000ff09f7d44a010200ffff0103dd00
[nfc.dev.transport] <<< 0000ff03fdd54b00e000
[nfc.dev.pn53x] poll NFC-F 00ffff0103
[nfc.dev.pn53x] InListPassiveTarget command with timeout 1000 ms
[nfc.dev.transport] >>> 0000ff09f7d44a010100ffff0103de00
[nfc.dev.transport] <<< 0000ff03fdd54b00e000
[nfc.dev.pn53x] RFConfiguration command with timeout 100 ms
[nfc.dev.transport] >>> 0000ff04fcd4320100f900
[nfc.dev.transport] <<< 0000ff02fed533f800
[nfc.dev.pn53x] polling for NFC-A technology

And on and on for a long time. 
If you want I can post a longer log...

nehpetsde, 2016-04-26:
(by stephen-tiedemann)
Sorry for confusing the command line parameters.

I would be interested in the log output when you touch the tag, from the last line that says "InListPassiveTarget" until the "Format error during RF communication" that appeared in your first log sample (with helloworld).

nehpetsde, 2016-04-26:
(by matthieu-falce)
I don't understand what you mean. 
When I do the following command : 
pi@raspberrypi ~/nfcpy $ python examples/helloworld.py

I have got the same log as before (the first one I posted). 

Also, I noticed that I have to launch a nfc-poll from the nfclib in order to initialize the device the first time.

nehpetsde, 2016-04-26:
(by thekarlmiller)
I am having the same problem.  Did anybody figure this out?

I have verified beyond a normal person's reasonable doubt that my solder joints are good, the wiring is correct, and I have followed the instructions (with no build errors) for libnfc (after first updating the RPi to the latest).

The nfc-poll seemed to work very intermittently for the first day, and for some reason, it seems to work every time now (go figure).

Anyway, it works, and I start it by doing a cold boot of the RPi, and executing these lines:
    cd /home/pi/libnfc/libnfc-1.7.0-rc7
    examples/nfc-poll
... which shows this:
     NFC reader: pn532_uart:/dev/ttyAMA0 opened
     NFC device will poll during 30000 ms (20 pollings of 300 ms for 5 modulations)
    ISO/IEC 14443A (106 kbps) target:
        ATQA (SENS_RES): 00  04
           UID (NFCID1): 0a  4e  a2  9b
          SAK (SEL_RES): 08

Then, I do this:    
     cd /root/nfcpy
     python examples/helloworld.py 
Here is my helloworld.py output with the connection to "tty:AMA0:pn53x" in helloworld.py:
    INFO:nfc.clf:searching for reader with path 'tty:AMA0:pn53x'
    INFO:nfc.clf:using NXP PN532 at /dev/ttyAMA0
    Please touch a tag to send a hello to the world

...as soon as I touch the tag (tried all six different tags Adafruit has for this type of reader and they all do the same thing), I see this:
    WARNING:nfc.dev:Driver.set_communication_mode() should be implemented.
    WARNING:nfc.dev.pn53x:[PN53x Error 0x13] Format error during RF communication
    WARNING:nfc.dev:Driver.set_communication_mode() should be implemented.
    WARNING:nfc.dev.pn53x:[PN53x Error 0x13] Format error during RF communication
    WARNING:nfc.dev:Driver.set_communication_mode() should be implemented.
    WARNING:nfc.dev.pn53x:[PN53x Error 0x13] Format error during RF communication
    WARNING:nfc.dev:Driver.set_communication_mode() should be implemented.
    WARNING:nfc.dev.pn53x:[PN53x Error 0x13] Format error during RF communication
    ...

When I try to run the tag-tool, whit the following command line:
     python examples/tagtool.py --device tty:AMA0:pn53x -d nfc.dev show
 ...I see this:
[main] enable debug output for module 'nfc.dev'
[nfc.clf] searching for reader with path 'tty:AMA0:pn53x'
[nfc.dev] trying pn53x on 'tty:AMA0:pn53x'
[nfc.dev] import nfc.dev.pn53x
[nfc.dev.transport] >>> 0000ff00ff00
[nfc.dev.pn53x] GetFirmwareVersion command with timeout 100 ms
[nfc.dev.transport] >>> 0000ff02fed4022a00
[nfc.dev.transport] <<< 0000ff06fad50332010607e800
[nfc.dev.pn53x] chipset is a PN532 version 1.6
[nfc.dev.pn53x] Diagnose command with timeout 100 ms
[nfc.dev.transport] >>> 0000ff08f8d400006e666370790c00
[nfc.dev.transport] <<< 0000ff08f8d501006e666370790a00
[nfc.dev.pn53x] ATR_RES timeout:   102.4 ms
[nfc.dev.pn53x] non-DEP timeout:    51.2 ms
[nfc.dev.pn53x] RFConfiguration command with timeout 100 ms
[nfc.dev.transport] >>> 0000ff06fad432020b0b0ad800
[nfc.dev.transport] <<< 0000ff02fed533f800
[nfc.dev.pn53x] set retries: ATR_REQ=2 PSL_REQ=1 PassiveTarget=3
[nfc.dev.pn53x] RFConfiguration command with timeout 100 ms
[nfc.dev.transport] >>> 0000ff06fad43205020103ef00
[nfc.dev.transport] <<< 0000ff02fed533f800
[nfc.clf] using NXP PN532 at /dev/ttyAMA0
[main] touch a tag
[nfc.dev.pn53x] polling for NFC-A technology
[nfc.dev.pn53x] InListPassiveTarget command with timeout 1000 ms
[nfc.dev.transport] >>> 0000ff04fcd44a0100e100
[nfc.dev.transport] <<< 0000ff03fdd54b00e000
[nfc.dev.pn53x] InListPassiveTarget command with timeout 1000 ms
[nfc.dev.transport] >>> 0000ff04fcd44a0104dd00
[nfc.dev.transport] <<< 0000ff03fdd54b00e000
[... 46 lines cut ...]
[nfc.dev.pn53x] [PN53x Error 0x13] Format error during RF communication
[nfc.dev.pn53x] polling for NFC-A technology
[nfc.dev.pn53x] InListPassiveTarget command with timeout 1000 ms
[nfc.dev.transport] >>> 0000ff04fcd44a0100e100
[nfc.dev.transport] <<< 0000ff0cf4d54b0101000408040a4ea29b3900
[nfc.dev.pn53x] found NFC-A target @ 106 kbps
[nfc.dev] Driver.set_communication_mode() should be implemented.
[nfc.dev.pn53x] RFConfiguration command with timeout 100 ms
[nfc.dev.transport] >>> 0000ff06fad432020b0b0bd700
[nfc.dev.transport] <<< 0000ff02fed533f800
[nfc.dev.pn53x] RFConfiguration command with timeout 100 ms
[nfc.dev.transport] >>> 0000ff04fcd4320403f300
[nfc.dev.transport] <<< 0000ff02fed533f800
[nfc.dev.pn53x] InDataExchange command with timeout 406 ms
[nfc.dev.transport] >>> 0000ff05fbd440013000bb00
[nfc.dev.transport] <<< 0000ff03fdd54113d700
[nfc.dev.pn53x] [PN53x Error 0x13] Format error during RF communication
...

and it keeps going like that.  

Any help would be greatly appreciated, as I need to get an example going for a Makerspace exhibit at a festival in Syracuse within the next day and a half.

Thanks so much,
Karl Miller - <email address hidden>

nehpetsde, 2016-04-26:
(by stephen-tiedemann)
Thanks for the detailed description. Apparently you have a Mifare 1K Tag which is not supported by nfcpy. Mifare 1K belongs to the Mifare Classic family of tags and requires a reader that supports the proprietary Mifare Crypto. While naturally this is built into NXP reader IC's it's not commonly supported by other chipsets. Mifare Classic is also not an NFC Forum Tag Type and nfcpy's main goal is to implement NFC Forum functionality. You would need to look for Mifare Ultralight (NFC Forum Type 2) or Smart MX (Type 4) Tags from NXP's offers. Other possible types are Topaz (Type 1)  from Broadcom and FeliCa (Type 3) from Sony.

nehpetsde, 2016-04-26:
(by 4barbes)
This also affect Mifaire Ultralight of 1ko

nehpetsde, 2016-04-26:
(by stephen-tiedemann)
Is this a MIFARE Ultralight EV1 (MF0ULx1) ?

nehpetsde, 2016-04-26:
(by stephen-tiedemann)
There's no plan to support Mifare Classic Tags in nfcpy.

closed 2016-04-26 by nehpetsde

kafffee, 2018-08-23:
So is it possible to read a simple NDEF-textstring from a tag with tagtool.py or the CLI? If so, how do I have to change it, respectively what command(s) do I use? 
And then, just print the NDEF text.
Im planning to use a pn532 or rc522 via SPI on a Raspberry Pi3 Model B..

nehpetsde, 2018-08-23:
Unless it is a Mifare Classic Tag, as was the cause for this issue, reading NDEF should work just fine.

kafffee, 2018-08-23:
Can you give me a short code example to, lets say, print the UID and NDEF text?

nehpetsde, 2018-08-24:
https://nfcpy.readthedocs.io/en/latest/topics/get-started.html#read-and-write-tags

https://nfcpy.readthedocs.io/en/latest/modules/tag.html#nfc.tag.Tag.identifier

kafffee, 2018-08-24:
First let me thank you for the quick response.

I tried to write some code but I think there might be some mistakes to it. Would you do me a favor and take a look at the following:

```
import nfc

class nfc.tag.Tag(clf, target)
class NDEF(tag)

while tag.ndef.is_present:
	if tag.ndef is not None:
		U_I_D = tag.ndef.identifier()
		print("The tag's UID is: " + U_I_D)
		if tag.ndef.is_readable() == True:
			for record in tag.ndef.records:
			print("The tag's NDEF text is: " + record)
		else:
			print("NDEF-text not readable.")
```

I'm a newbie so excuse my questions. Before I get myself a nfc scanner module I would like to know if it is even makeable. Will this work for a RC522 or PN532 tag scanner or will it only work with a USB device?

nehpetsde, 2018-08-28:
A PN532 based reader should work just fine but I don't know about RC522. The documentation has sufficient code examples to get you going. There's also a section about well supported and tested reader devices.

kafffee, 2018-08-29:
You say PN532-based. I found boards from 6 to 40 dollars, from adafruit to NXP/itead amongst others. In the table from the nfcpydocs there was only the one from adafruit listed (v1.6). Can you give me a recommendation that would meet my needs? They all looked fairly different and also had lots of jumper contacts, I was wondering because I would use UART without USB and that would only require 4 right? 
Thank you for the hints concerning the software though I was able to find all the information I needed on nfcpy.readthedocs.io. Couldnt test it out yet though because I dont have the board yet. 
It would be helpful to get a board with short delivery time.

nehpetsde, 2018-08-30:
Here's a thread on the ITEAD module: https://answers.launchpad.net/nfcpy/+question/258474

And this https://www.elechouse.com/elechouse/index.php?main_page=product_info&cPath=90_93&products_id=2242 I've used myself.

kafffee, 2018-08-30:
Thank you for this. I think I'm going with the PN532 NFC/RFID controller breakout board from Adafruit as the other ones seem to have low reading/writing range or the pins might break out and I dont want to solder anything. The issue is that I want to use it through a single-walled glas door (thickness 2-3millimeters) and a plastic casing, which might take couple of millimeters, too. Also, does anyone know what nfc tag and card I could use for this?

kafffee, 2018-08-30:
edit: The vendor I found didnt provide any information about the reading/writing range of the adafruit breakout board so did anyone make any experiences about the range?

kafffee, 2018-09-14:
So I got myself the Adafruit PN532 RFID NFC controller breakout board, done some soldering and hooked it up to my Raspberry via UART and installed nfcpy. The smd power led on the reader card is lit. So far so good. 
When I try to test it with “python -m nfc“ i get the output:
No  handlers could be found for logger "nfc.llcp.sec"
This is the 0.13.5 version of nfcpy run in Python 2.7.13
on Linux-4.9.76-v7+-armv7l-with-debian-9.3
I'm now searching your system for contactless devices
I'm not trying serial devices because you haven't told me
-- add the option '--search-tty' to have me looking
-- but beware that this may break other serial devs
Sorry, but I couldn't find any contactless device
I assume that I did something wrong configuring, or that I use the wrong version of python. Is there still no support for python 3 and how can I find out what version I am currently running??

nehpetsde, 2018-09-14:
If you do not use a USB dongle then it should used as a serial device. Please read the command output carefully, it tells you almost all you asked for. Remaining only: no Python 3 support yet.

kafffee, 2018-09-15:
Well I'm a beginner so this leaves some questions: First of all I do not use a usb dongle but connected the.board with 4 pins directly via UART. What do you mean by “it should be used as serial device. “
I tried to add “--search-tty“, too, and the program seems to be doing something but after minutes, theres no output.
What can I do in.easy steps?

kafffee, 2018-09-18:
A little PostScript: while I was tryin to get things working i checked my wiring. The 5V and GND seems to be properly connected as the power led on the breakout board is on.
One Website I found told me to hook up the RX pin of the breakout board with the UART RX port of the Raspberry (GPIO15) and TX with (GPIO14). The other websíte told me to do it the other way around. Which one is the right way?
Other than that, there might be a bad soldering Joint.
However there is no contactless device detected.
Please help

kafffee, 2018-09-25:
Hey Stephen,

As you did not answer yet, I assume you do not have an answer to my questions.

So what do you think about just using SPI instead? Can I just use tty:AMA0:pn532 as path.argument for nfc.clf.ContactlessFrontend.open () as well?