nfcpy/nfcpy #40: IOError: [Errno 19] No such device

nehpetsde, 2016-04-26:
I'm occasionally seen this error message on my R-Pi... How can I prevent it or fix it?

Traceback (most recent call last):
  File "mainNfc.py", line 221, in <module>
    clf = nfc.ContactlessFrontend('tty:AMA0:pn532')
  File "/home/pi/nfcpy/nfc/clf/__init__.py", line 72, in __init__
    if path and not self.open(path):
  File "/home/pi/nfcpy/nfc/clf/__init__.py", line 146, in open
    self.device = device.connect(path)
  File "/home/pi/nfcpy/nfc/clf/device.py", line 104, in connect
    device = driver.init(tty)
  File "/home/pi/nfcpy/nfc/clf/pn532.py", line 368, in init
    raise IOError(errno.ENODEV, os.strerror(errno.ENODEV))
IOError: [Errno 19] No such device

Thanks,

-Juan

nehpetsde, 2016-04-26:
(by stephen-tiedemann)
Can you try with latest trunk version, there is at least this http://bazaar.launchpad.net/~stephen-tiedemann/nfcpy/trunk/revision/267 change in relation to pn532 on serial port.

nehpetsde, 2016-04-26:
(by juancarlos-trujillo)
Ok. I'm going to ask my team to update the stack to see if this problem goes away.

nehpetsde, 2016-04-26:
(by juancarlos-trujillo)
We have definitely seen this error 4 times so far during this month

jctlmm, 2016-06-29:
FYI, we are now using 0.11.0 and haven't seen the error so far...

PanderMusubi, 2016-09-27:
This error showed up on Raspberry Pi with Raspbian recently while it worked before. We need it to be functional this Thursday so we are a bit caught by surprise. the device we use is:

    # lsusb
    Bus 001 Device 008: ID 072f:2200 Advanced Card Systems, Ltd
    ...

What are the possibilities we have to patch this quickly? Any help is much appreciated.

PanderMusubi, 2016-09-27:
Any help on how I can debug this is very welcome.

PanderMusubi, 2016-09-27:
The program we test with is

    #!/usr/bin/env python
    
    from nfc import ContactlessFrontend
    from time import sleep
              
    
    def connected(tag):
        ident = ''.join('{:02x}'.format(ord(c)) for c in tag.identifier)
        print(ident)
        return False
        
    clf = ContactlessFrontend('usb')
    while True:
        clf.connect(rdwr={'on-connect': connected})
        sleep(1)

On latest Raspbian with RPi2.

PanderMusubi, 2016-09-28:
Fix is to blacklist kernel module and add user to proper group, see documentation.

nehpetsde, 2017-03-27:
This question seems to be resolved.

closed 2017-03-27 by nehpetsde

nolansingroy, 2021-04-07:
how do you blacklist the kernel module? where in the documentation is the answer to this?

nolansingroy, 2021-04-07:
I am on raspi lite PI 4 with a Wave share Pn532 and receiving error no 19 when I attach a piezo component to gpio 24 all the ting and 25 some times?