nfcpy/nfcpy #45: input/output error while waiting for ack issue after 1st successful run

prattom, 2016-06-16:
I have a board with i.mx6 UL processor and running yocto linux. Whenever I try out the following command to read tag 
python tagtool.py -v --device tty:mxc1:pn532 show
it runs successfully but after reading tag it shows error
[nfc.clf.pn532] input/output error while waiting for ack
and after this error the above mentioned command fails every time with following error
no contact less reader available 
every time till we reset the pn532 module power.

Running following command on successful run (1st time after power reset)
python tagtool.py -d nfc.tag --reltime -v --device tty:mxc1:pn532

shows following output 

1360 ms [main] enable debug output for 'nfc.tag'
1362 ms [nfc.clf] searching for reader on path tty:mxc1:pn532
1633 ms [nfc.clf] using PN532v1.6 at /dev/ttymxc1
** waiting for a tag **
8002 ms [nfc.tag] trying to activate 106A sdd_res=E63AE5D5 sel_res=08 sens_res=0400
8004 ms [nfc.tag] trying type 2 tag activation for 106A
Type2Tag ID=E63AE5D5
8016 ms [nfc.tag.tt2] read ndef data
8018 ms [nfc.tag.tt2] read pages 0 to 3
8020 ms [nfc.tag.tt2] >> 3000 (0.005000s)
8049 ms [nfc.tag.tt2] << 04 (0.027840s)
8051 ms [nfc.tag.tt2] received nak response
8092 ms [nfc.tag.tt2] first four memory pages were unreadable
Memory Dump:
8094 ms [nfc.tag.tt2] read pages 0 to 3
8096 ms [nfc.tag.tt2] >> 3000 (0.005000s)
8129 ms [nfc.tag.tt2] << 04 (0.031544s)
8131 ms [nfc.tag.tt2] received nak response
8172 ms [nfc.tag.tt2] read pages 1 to 4
8174 ms [nfc.tag.tt2] >> 3001 (0.005000s)
8209 ms [nfc.tag.tt2] << 04 (0.032929s)
8211 ms [nfc.tag.tt2] received nak response
8252 ms [nfc.tag.tt2] read pages 2 to 5
8254 ms [nfc.tag.tt2] >> 3002 (0.005000s)
8289 ms [nfc.tag.tt2] << 04 (0.033529s)
8291 ms [nfc.tag.tt2] received nak response
8332 ms [nfc.tag.tt2] read pages 3 to 6
8334 ms [nfc.tag.tt2] >> 3003 (0.005000s)
8369 ms [nfc.tag.tt2] << 04 (0.033435s)
8371 ms [nfc.tag.tt2] received nak response
8412 ms [nfc.tag.tt2] read pages 4 to 7
8414 ms [nfc.tag.tt2] >> 3004 (0.005000s)
8449 ms [nfc.tag.tt2] << 04 (0.033399s)
8451 ms [nfc.tag.tt2] received nak response
  000: ?? ?? ?? ?? (UID0-UID2, BCC0)
  001: ?? ?? ?? ?? (UID3-UID6)
  002: ?? ?? ?? ?? (BCC1, INT, LOCK0-LOCK1)
  003: ?? ?? ?? ?? (OTP0-OTP3)
8603 ms [nfc.clf.pn532] input/output error while waiting for ack

While same pn532 module work ok with normal pc every time without any error.

nehpetsde, 2016-07-04:
I would be guessing that with a normal PC you're using an FTDI friend to connect the serial line through USB. In my experience (with RPi) it is then behaving just a little different about timings. One change committed last year Nov 25 was specifically about ack frame handling with PN532, could you check whether you're on a version including that change (0.11.x would).

AndiCui, 2016-08-22:
This is still problematic on a raspberry pi (the latest from GITHUB)
```
Traceback (most recent call last):
  File "tap4id.py", line 5, in <module>
    clf = nfc.ContactlessFrontend('tty:USB0:pn532')
  File "build/bdist.linux-armv7l/egg/nfc/clf/__init__.py", line 71, in __init__
  File "build/bdist.linux-armv7l/egg/nfc/clf/__init__.py", line 145, in open
  File "build/bdist.linux-armv7l/egg/nfc/clf/device.py", line 104, in connect
  File "build/bdist.linux-armv7l/egg/nfc/clf/pn532.py", line 366, in init
  File "build/bdist.linux-armv7l/egg/nfc/clf/transport.py", line 134, in read
IOError: [Errno 110] Connection timed out
```
How can I help to debug?

nehpetsde, 2016-08-24:
It works flawless on my RPi 2, tested with both 0.11.1 from PyPI and github master. My RPi is: 
$ uname -a
Linux mico 4.4.19-v7+ #906 SMP Tue Aug 23 15:53:06 BST 2016 armv7l GNU/Linux
$ vcgencmd version
Aug 23 2016 15:55:25
Copyright (c) 2012 Broadcom
version f23e65aeeffca65654550c93254a0ee55f34fb07 (clean) (release)

May that be a difference? I know about issues with the Broadcom USB on RPi 1, not sure if that could be the reason.

AndiCui, 2016-08-24:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.4.13-v7+ #894 SMP Mon Jun 13 13:13:27 BST 2016 armv7l GNU/Linux
pi@raspberrypi:~ $ vcgencmd version
Jun 20 2016 18:13:27 
Copyright (c) 2012 Broadcom
version 460b39c674ef2ac89360e15331f97a2cb1388a44 (clean) (release)
pi@raspberrypi:~ $ 
```
I am on RPi 3. It doesnt always get the problem, but it happens really often. Avg 3/5 times.

nehpetsde, 2016-08-30:
Unfortunately I can now only confirm that there is erratic behaviour with PN532 connected through USB converter with both RPi 2 and 3. I've run numerous repetitions of the tagtool.py script with different sleep times between invocations but there's no clear pattern to observe. The same tests on a standard PC just always work and there's also no difference between PN532v1.6 and v1.4. Maybe you could try on yours with varying sleep times:

for (( i=1; i <= 10; i++ )); do examples/tagtool.py --device tty:USB0:pn532; sleep 1; done

nehpetsde, 2016-09-01:
At least I've found a pattern but it's not very useful yet. Maybe someone reading this can help.

On RPi 2 and 3, with a PN532 connected through /dev/ttyUSB0 I'm running the shortest possible sequence of just open the serial port, send a GetVersion command and read the acknowledge, then close the port. When I call this script in a shell loop as above, slightly changed to provide units of 0.1 seconds wait time, there is a clear pattern of times from 0 to 1 second good behaviour, then from 1 to 2 seconds erroneous behaviour and this repeats when going up (I stop at 10 seconds). And similar outcome with a small C program, so it's not Python and not the python-serial module.

nehpetsde, 2016-09-08:
After quite some testing and digging I strongly believe that the root cause lies in the USB Host Controller Driver or the USB hardware implementation. A lengthy write-up of the journey can be found [here](https://github.com/nfcpy/nfcpy/wiki/USB-TTL-serial-adapter-on-Raspberry-Pi). While the problem itself might persist, it is possible for nfcpy to work around it by extending the timeout value for the very first read from PN532. The drawback is a variable additional time of up to 1000 ms from program start to NFC exchanges when using a USB serial adapter on Raspberry Pi. I don't think this will be an issue for the majority of applications, but for a sort of assembling line application it might be better to use the RasPi's UART.

commit 8ba5356d (2016-09-08): Work-around an issue with the USB host implementation on Raspberry Pi. This resolves #45 with a longer timeout for the for the very first read from a PN532 through a USB serial adapter on Raspberry Pi. The background is documented at https://github.com/nfcpy/nfcpy/wiki/USB-TTL-serial-adapter-on-Raspberry-Pi
closed 2016-09-08 by nehpetsde