nfcpy/nfcpy #46: Does nfcpy always polls when waiting for a NFC tag?

jctlmm, 2016-07-01:
We use the following code to wait for a NFC tag and we are unsure how nfcpy handles this underneath...
```
import nfc

def connected(tag):
    """ Print the NFC Tag information"""

clf = nfc.ContactlessFrontend('tty:AMA0:pn532')

timeOut = lambda: time.time() - started > (5 * 60) 
started = time.time()
nfcTag = clf.connect(rdwr={'on-connect': connected}, terminate=timeOut)

""" Do something with the tag here"""
```
We are concern that nfcpy always poll waiting for a tag, thus the power consumption on our device is not optional.

Any feedback will be really appreciated...

nehpetsde, 2016-07-04:
It is indeed true that clf.connect() continuously polls for a tag, after all there is no other way to find one. The default polling strategy is to sense the set of rdwr['targets'] for 5 times with an interval of 500ms and then check the terminate condition. This continues until terminate becomes True. During the 500ms (minus the time it was polling) there is be no RF field generated.

The polling strategy is set through the iterations and interval options that clf.connect() uses when calling clf.sense(). Thus one way to influence polling is to use clf.sense() API from application code. Other ways are to either add delay in the terminate callback function (that's of course a bit of a hack), or patch the nfc/clf/__init__.py file to use different values.

Finally, since a couple of minutes the version on the git master branch will allow the 'iterations' and 'interval' to be set as rdwr options.

jctlmm, 2016-07-05:
Since our use case really cares about power consumption, we would expect that the HW may be able to generate an interrupt when a new tag is detected, instead of polling from SW. 

So, are you saying that latest version of nfcpy will allow us to change the values of 'iterations' and 'interval'? Having this possibility seems to me a good idea so we can fine tune the algorithm from our side... If this is what is going to be included on the latest version, would you mind to share an example on how can we do it?

We would like to avoid patching files... that way we don't deviate too much from the official nfcpy releases.

nehpetsde, 2016-07-06:
Both parameters are documented for the latest version of nfcpy at https://nfcpy.readthedocs.io/en/latest/modules/clf.html#nfc.clf.ContactlessFrontend.connect. In a nutshell, if you use connect(rdwr={..., 'iterations': 5, 'interval': 0.5}) that'd be equivalent to the default settings. Note that a new iterations cycle is always started after the terminate callback (except the first time). This means that you could also just set iterations to 1 and implement a waiting time in the terminate function, possibly going into a deeper sleep state for that time.

jctlmm, 2016-07-07:
Thanks for the detailed information. We'll definitely give it a try...

nehpetsde, 2017-03-27:
This question seems to be resolved.

closed 2017-03-27 by nehpetsde