nfcpy/nfcpy #48: WindowsError: [Error 126] when import nfc in nfcpy 0.11.1

ghxbob, 2016-08-02:
everything is fine with nfcpy 0.10.0, however when i use the latest release nfcpy 0.11.1, i got the following error.

D:\>cd nfcpy-0.11.1

D:\nfcpy-0.11.1>python
Python 2.7.9 (default, Dec 10 2014, 12:24:55) [MSC v.1500 32 bit (Intel)] on w
32
Type "help", "copyright", "credits" or "license" for more information.
>>> import nfc
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "nfc\__init__.py", line 28, in <module>
    from clf import ContactlessFrontend
  File "nfc\clf\__init__.py", line 35, in <module>
    from . import device
  File "nfc\clf\device.py", line 39, in <module>
    from . import transport
  File "nfc\clf\transport.py", line 32, in <module>
    import usb1 as libusb
  File "C:\Python27\lib\site-packages\usb1.py", line 60, in <module>
    import libusb1
  File "C:\Python27\lib\site-packages\libusb1.py", line 193, in <module>
    libusb = _loadLibrary()
  File "C:\Python27\lib\site-packages\libusb1.py", line 167, in _loadLibrary
    return dll_loader('libusb-1.0' + suffix, **loader_kw)
  File "C:\Python27\lib\ctypes\__init__.py", line 365, in __init__
    self._handle = _dlopen(self._name, mode)
WindowsError: [Error 126]
>>>

I'm quite sure everything works well if i switch to nfcpy 0.10.0, see the below traces.
D:\nfcpy-0.10.0>python
Python 2.7.9 (default, Dec 10 2014, 12:24:55) [MSC v.1500 32 bit (Intel)] on win
32
Type "help", "copyright", "credits" or "license" for more information.
>>> import nfc
>>> clf = nfc.ContactlessFrontend('usb')
>>> print clf
SONY RC-S380/S on usb:000:001
>>> exit()

Skylled, 2016-08-02:
Looks like you don't have libusb-1.0 installed on your computer. Put this (MS32 or MS64) .dll into either your script's folder or in system32, https://sourceforge.net/projects/libusb/files/libusb-1.0/libusb-1.0.20/libusb-1.0.20.7z/download

ghxbob, 2016-08-03:
Hi Skylled,
Thank for your comment, after i copy libusb-1.0.dll into system32 folder, i met a new problem.
And if i run with nfcpy-0.9.2, everything works fine.(see below traces)
Hope you have a further look at this problem.

D:\nfcpy-0.11.1>python
Python 2.7.9 (default, Dec 10 2014, 12:24:55) [MSC v.1500 32 bit (Intel)] on win
32
Type "help", "copyright", "credits" or "license" for more information.
>>> import nfc
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "nfc\__init__.py", line 28, in <module>
    from clf import ContactlessFrontend
  File "nfc\clf\__init__.py", line 35, in <module>
    from . import device
  File "nfc\clf\device.py", line 39, in <module>
    from . import transport
  File "nfc\clf\transport.py", line 37, in <module>
    import serial
  File "C:\Python27\lib\site-packages\serial\__init__.py", line 27, in <module>
    from serial.serialwin32 import Serial
  File "C:\Python27\lib\site-packages\serial\serialwin32.py", line 15, in <modul
e>
    from serial import win32
  File "C:\Python27\lib\site-packages\serial\win32.py", line 182, in <module>
    CancelIoEx = _stdcall_libraries['kernel32'].CancelIoEx
  File "C:\Python27\lib\ctypes\__init__.py", line 378, in __getattr__
    func = self.__getitem__(name)
  File "C:\Python27\lib\ctypes\__init__.py", line 383, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: function 'CancelIoEx' not found

run with nfcpy-0.9.2, everything works fine.

D:\nfcpy-0.9.2>python
Python 2.7.9 (default, Dec 10 2014, 12:24:55) [MSC v.1500 32 bit (Intel)] on win
32
Type "help", "copyright", "credits" or "license" for more information.
>>> import nfc

Skylled, 2016-08-03:
Okay so now the problem appears to be that the 'serial' package is calling for a function that wasn't introduced until Windows Vista. I'm guessing that you're running on Windows XP or earlier. If that's the case, there's two solutions that come to mind, but feel free to seek a second opinion.
Option A: Stay on NFCpy 0.10 or earlier.
Option B: Make some adjustments to the code to remove the serial feature. (This assumes you're exclusively using USB devices, which I know the RC-S380 is.)

ghxbob, 2016-08-05:
Hi Skylled,

Thanks for your support, i'd like to stay on 0.10.0, I hope it is a good start point.

closed 2016-08-05 by ghxbob