nfcpy/nfcpy #49: Problem with the llcp

jinzhangA, 2016-09-16:
Dear Stephen:
I am trying to use the llcp funtion of your library. I used your sample code to setup one side of the nfc. However, this only return some error. Can you take a look for me?

`import nfc
import nfc.llcp
def server(socket):
    # note that this server only accepts one connection
    # for multiple connections spawn a thread per accept
    while True:
       client = socket.accept()
       while True:
           message = client.recv()
           print("Client said: {0}".format(message))
           client.send("It's me!")

def client(socket):
    socket.connect( 'urn:nfc:xsn:nfcpy.org:test-service' )
    socket.send("Hi there!")
    message = socket.recv()
    print("Server said: {0}".format(message))

def startup(clf,llc):
    socket = nfc.llcp.Socket(llc, nfc.llcp.DATA_LINK_CONNECTION)
    socket.bind( 'urn:nfc:xsn:nfcpy.org:test-service' )
    print("server bound to SAP {0}".format(socket.getsockname()))
    socket.listen()
    Thread(target=server, args=(socket,)).start()
    return llc

def connected(llc):
    socket = nfc.llcp.Socket(llc, nfc.llcp.DATA_LINK_CONNECTION)
    Thread(target=client, args=(socket,)).start()
    return True

clf = nfc.ContactlessFrontend('tty:AMA0:pn532')
clf.connect(llcp={'on-startup': startup, 'on-connect': connected})`

`INFO:nfc.clf:searching for reader on path tty:AMA0:pn532
INFO:nfc.clf:using PN532v1.6 at /dev/ttyAMA0
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-8-f4bcabd49a76> in <module>()
     31 
     32 clf = nfc.ContactlessFrontend('tty:AMA0:pn532')
---> 33 clf.connect(llcp={'on-startup': startup, 'on-connect': connected})

/usr/local/lib/python2.7/dist-packages/nfcpy-latest-py2.7.egg/nfc/clf/__init__.pyc in connect(self, **options)
    504 
    505             llc = nfc.llcp.llc.LogicalLinkController(**llcp_options)
--> 506             llc = llcp_options['on-startup'](llc)
    507             if isinstance(llc, nfc.llcp.llc.LogicalLinkController):
    508                 llcp_options['llc'] = llc

TypeError: startup() takes exactly 2 arguments (1 given)
`

nehpetsde, 2016-09-17:
That's a documentation error. The startup() function is called with only the llc argument. I'll correct the example, sorry 'bout that.

jinzhangA, 2016-09-17:
Thank you, Steven! There's another typo of in the doc,
def client(socket):
    socket.sendto("message", address=16)

sento takes addr, not address.

nehpetsde, 2016-09-19:
Thanks a lot, it's now corrected.

closed 2017-03-27 by nehpetsde