nfcpy/nfcpy #51: Need help with LLCP!!!

jinzhangA, 2016-09-23:
I am trying to make two Raspberry Pi talk though llpc.
At one end:
import nfc
import nfc.llcp
import threading

def server(socket):
    message, address = socket.recvfrom()
    socket.sendto("It's me!", address)
    socket.close()

def on_startup(llc):
    socket = nfc.llcp.Socket(llc, nfc.llcp.LOGICAL_DATA_LINK)
    socket.bind(address=32)
    threading.Thread(target=server, args=(socket,)).start()
    return llc

llcp_options = {
    'on-startup': on_startup,
    #'on-connect': on_connect,
}
with nfc.ContactlessFrontend('tty:AMA0:pn532') as clf:
    clf.connect(llcp=llcp_options)
    print("link terminated")

The other end:
import nfc
import nfc.llcp
import threading

def client(socket):
    socket.sendto("Hi there!", address=32)
    socket.close()

def on_connect(llc):
    socket = nfc.llcp.Socket(llc, nfc.llcp.LOGICAL_DATA_LINK)
    threading.Thread(target=client, args=(socket,)).start()
    return True

llcp_options = {
#     'on-startup': on_startup,
    'on-connect': on_connect,
}
with nfc.ContactlessFrontend('tty:AMA0:pn532') as clf:
    clf.connect(llcp=llcp_options)
    print("link terminated")

This may work for the first time after reboot, but after that:
link terminated
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "caonima.py", line 6, in server
    message, address = socket.recvfrom()
  File "build/bdist.linux-armv7l/egg/nfc/llcp/socket.py", line 141, in recvfrom
    return self.llc.recvfrom(self._tco)
  File "build/bdist.linux-armv7l/egg/nfc/llcp/llc.py", line 819, in recvfrom
    return socket.recvfrom()
  File "build/bdist.linux-armv7l/egg/nfc/llcp/tco.py", line 296, in recvfrom
    raise err.Error(errno.EPIPE)
Error: nfc.llcp.Error: [EPIPE] Broken pipe

Really need some help here!

nehpetsde, 2016-09-23:
Could you please try with "examples/beam.py send ..." and "examples/beam.py recv ..." on the both sides.

jinzhangA, 2016-09-23:
Just tested the command you suggest.
One one side: python beam.py --device='tty:AMA0:pn532' recv print
The other side:  python beam.py --device='tty:AMA0:pn532' send text '123'

These two commands can work as the llpc ones. However, once I remove the NFC board from each other without 'ctrl+c' first, running these commands again will always say 'broken pipe'. How to fix or avoid this?

jinzhangA, 2016-09-23:
[image]

[image]

jinzhangA, 2016-09-23:
After some tests, I just found out the LLCP with my setup can only work at the first time after system reboot. Any idea how to fix this?

nehpetsde, 2016-09-23:
Have you tried with latest version from github master branch? RPi 1 or 2? Adafruit PN532 board or Elechouse?

Further the debug messages may help to understand. For beam.py that's with the `-d nfc.llcp`, `-d nfc.dep` and `nfc.clf` options, `-f <file>` to write to file, `--reltime` to look after timing issues.

jinzhangA, 2016-09-23:
I cloned the master branch and installed it by 'sudo python setup.py install'. I guess this would give me the latest version of nfcpy. I am using Adafruit PN532 board

closed 2017-03-27 by nehpetsde