nfcpy/nfcpy #53: Card emulation not working again

jinzhangA, 2016-09-23:
Since I failed to setup the LLPC, I switched back to the card emulation method. However, the card emulation seems not working again. I am using adafruit pn532 with Raspi 2

import nfc

sensf_res = bytearray.fromhex('01 03FEFFE011223344 01E0000000FFFF00 12FC')
def on_startup(target):
    target.brty = "212F"
    target.sensf_res = sensf_res
    return target

clf = nfc.ContactlessFrontend('tty:AMA0:pn532')
clf.connect(rdwr={'on_startup': on_startup})

I tried running this code on Raspi, and read it with my Android Phone and another Pi as reader, neither of these works. 
Anyone has any idea? Really need some help here!

dh0271, 2016-11-12:
Running the example/tagtool file using emulate using both NFC Tools and TagWriter my Nexus 6P fails to read the 'tag'. Using adafruit pn532 with Raspi 3.

nehpetsde, 2016-11-14:
@jinzhangA - The card emulation options must be given as card={...} in clf.connect(). Please have a look at http://nfcpy.readthedocs.io/en/latest/topics/get-started.html#emulate-a-card example.

@dh0271 - If PN532 is connected through USB (via FTDI adapter) it should work. If connected straight to serial line the default connect string is "tty:S0:pn532", not AMA0 as for RPi2 (unless you used dtoverlay to switch AMA0 from Bluetooth back to the serial connector). I'd also recommend that youpply some NDEF initial content for the tag emulation "... emulate FILE.NDEF tt3 ...", that makes the phone able to read some data. You can create an NDEF file with the ndeftool.py script.

dh0271, 2016-11-14:
I am using the RPi3 and am using dtoverlay to switch AMA0 from Bluetooth back to serial. All the examples thus far have worked and have been successful in using beam as well as reading and writing to NFC cards. I have tried supplying initial content for the tag emulation but my phone does not recognize any NFC tag.

nehpetsde, 2016-11-15:
I've tried again all possible combinations: Adafruit PN532 on RPi 2 and 3 connected on UART-GPIO and USB-FTDI and for RPi 3 with ttyS0 (miniuart) and ttyAMA0 (dtoverlay=pi3-miniuart-bt). Used tagtool to emulate a Type 3 Tag with a small NDEF message. Used a Sony Xperia Z and a Google Nexus 5 to read. All worked fine.
```
$ examples/ndeftool make smartposter http:nfcpy.org -o nfcpy.ndef
$ example/tagtool.py --device tty:AMA0:pn532 emulate nfcpy.ndef tt3
```
Does your phone read other (physical) type 3 tags?

dh0271, 2016-11-20:
I tried the above commands tag emulate and my Nexus 6P did not recognize anything. I then used the commands
```
$ examples/ndeftool make smartposter http:nfcpy.org -o nfcpy.ndef
$ examples/tagtool.py --device tty:AMA0:pn532 load nfcpy.ndef
```
And that successfully wrote to a Desfire card which my phone was able to read.

I don't have any Type 3 tags available so I am not sure if the issue is the library, the reader or my phone.

nehpetsde, 2016-11-23:
I've got hold of a Nexus 6 (not 6P) and it's working when using nfcpy master branch version. It's not working with nfcpy-0.11.1 and maybe you are using that. You can find out the version that gets imported by adding "print(nfc.__version__) right after the import statements in tagtool.py.

To further isolate things you could use virtualenv:
```
$ virtualenv test-emulate
$ cd test-emulate
$ . bin/activate
$ pip install libusb1 pyserial
$ git clone https://github.com/nfcpy/nfcpy.git
$ cd nfcpy
$ examples/ndeftool make smartposter http:nfcpy.org -o nfcpy.ndef
$ examples/tagtool.py --device tty emulate nfcpy.ndef tt3
```

dh0271, 2016-11-23:
Using the master branch from GitHub works for me on the Nexus6P. I was using some tutorial somewhere that had me use bazaar to get it, possibly with the non master branch.

Thanks for the help.

closed 2016-12-15 by nehpetsde