nfcpy/nfcpy #54: rcs380.py: TypeError: format() argument after * must be a sequence, not NoneType

malikolivier, 2016-10-16:
Hello,

We've been using nfcpy on an Intel Edison with a USB-plugged Sony RC-S380 for quite some time now.
We are spawning the `examples/tagpool.py` script from our program like this:

```
python examples/tagpool.py -l
```

Then we listen to it's output and from then we get the value of the scanned ID card.
It works very fine, but sometimes the process crashes with the following error:

```
nfcpy/nfc/clf/rcs380.py", line 307, in get_firmware_version
   log.debug("firmware version {1:x}.{0:02x}".format(*data))
TypeError: format() argument after * must be a sequence, not NoneType
```

I was not able to reproduce this problem. However, it happens regularly on our production devices (and seemingly randomly), often enough so that we are worrying about it.

Does anyone have any idea where does this come from and how to fix it?
Thanks.

nehpetsde, 2016-10-20:
The tagtool's way of looping is by complete startup-shutdown cycle for the whole stack including device driver. In driver startup the very first chip communication with evaluated response is the get_firmware_version() command and there seems to be no response data. Why that happens is difficult to say without debug logs (maybe even with).

In general I would recommend for your type of application to avoid always opening and closing the driver but just loop over card detection. The examples/sense.py script follows this approach and is maybe already sufficient for just getting the card ID.

malikolivier, 2016-10-21:
Thanks! We'll try out your suggestions.

I'll be back with feedbacks later.

malikolivier, 2016-11-29:
I am back ;)

So we tried out your solution, and sure it works better with `sense.py`.

However, the bug I talked about above is still there.
```
root@edison:~# python nfcpy/examples/sense.py 212F -r
607 ms [nfc.clf] searching for reader on path usb
Traceback (most recent call last):
  File "external/nfcpy/examples/sense.py", line 165, in <module>
    main(parser.parse_args())
  File "external/nfcpy/examples/sense.py", line 50, in main
    if clf.open(args.device):
  File "/opt/sazabi/external/nfcpy-write_log/nfc/clf/__init__.py", line 145, in open
    self.device = device.connect(path)
  File "/opt/sazabi/external/nfcpy-write_log/nfc/clf/device.py", line 84, in connect
    device = driver.init(transport.USB(bus, dev))
  File "/opt/sazabi/external/nfcpy-write_log/nfc/clf/rcs380.py", line 882, in init
    chipset = Chipset(transport, logger=log)
  File "/opt/sazabi/external/nfcpy-write_log/nfc/clf/rcs380.py", line 181, in __init__
    self.get_firmware_version()
  File "/opt/sazabi/external/nfcpy-write_log/nfc/clf/rcs380.py", line 307, in get_firmware_version
    log.debug("firmware version {1:x}.{0:02x}".format(*data))
TypeError: format() argument after * must be a sequence, not NoneType
```

When the bug starts happening, it will always happen every time `sense.py` is run, unless we unplug and replug the NFC reader or reboot the whole system.

The bug never happens when `sense.py` is run for the first time since boot. This lets me think that `nfcpy` (or the linux kernel) somewhat did not end properly and still has a lock on the device.

Our kernel version is 3.10.17.

nehpetsde, 2016-12-02:
I'll try to reproduce but need more information. Which version of nfcpy are you using? In a Python shell do "import nfc" and "print(nfc.__version__)". How often (roughly) do you have to tap until the reader becomes dead? Is the reader the only thing connected to the Edison OTG plug?

malikolivier, 2016-12-06:
We are using the `latest` version of nfcpy (that's the output of `nfc.__version__`). To be precise, it is fixed at commit be17f7cee8ca396cbaeb5db9eb37d7ac88f78545.

I ran the script with the debug/verbose flags. Here is the output:

```
python nfcpy/examples/sense.py 212F -r --debug --verbose
605 ms [nfc.clf] searching for reader on path usb
606 ms [nfc.clf.transport] using libusb-1.0.19
618 ms [nfc.clf.transport] path matches '^(usb|)$'
634 ms [nfc.clf.device] loading rcs380 driver for usb:054c:06c3
656 ms [nfc.clf.transport] >>> 0000ff00ff00
658 ms [nfc.clf.rcs380] SetCommandType 01
660 ms [nfc.clf.transport] >>> 0000ffffff0300fdd62a01ff00
662 ms [nfc.clf.transport] <<< 0000ffffff0400fcd7211101f600
663 ms [nfc.clf.rcs380] GetFirmwareVersion 
664 ms [nfc.clf.transport] >>> 0000ffffff0200fed6200a00
666 ms [nfc.clf.transport] <<< 0000ff00ff00
668 ms [nfc.clf.transport] <<< 0000ff00ff00
Traceback (most recent call last):
  File "/opt/nfcpy/examples/sense.py", line 165, in <module>
    main(parser.parse_args())
  File "/opt/nfcpy/examples/sense.py", line 50, in main
    if clf.open(args.device):
  File "/opt/nfcpy-write_log/nfc/clf/__init__.py", line 145, in open
    self.device = device.connect(path)
  File "/opt/nfcpy-write_log/nfc/clf/device.py", line 84, in connect
    device = driver.init(transport.USB(bus, dev))
  File "/opt/nfcpy-write_log/nfc/clf/rcs380.py", line 882, in init
    chipset = Chipset(transport, logger=log)
  File "/opt/nfcpy-write_log/nfc/clf/rcs380.py", line 181, in __init__
    self.get_firmware_version()
  File "/opt/nfcpy-write_log/nfc/clf/rcs380.py", line 307, in get_firmware_version
    log.debug("firmware version {1:x}.{0:02x}".format(*data))
TypeError: format() argument after * must be a sequence, not NoneType
```

I am now running some experiments to determine accurately for reproducibility. I'll be back with more results later. We do have other things plugged to USB, via a USB hub. But the bug still occurs when the NFC is the only thing plugged to the USB hub.

malikolivier, 2016-12-06:
We got the issue reproduced.

On Edison we launched the following bash script:

```sh
count=0
while true; do
	# Start nfcpy and send SIGKILL to it after 2 seconds
	((count++))
	python nfcpy/examples/sense.py 212F -r &
	PID=$!
	echo "PID: $PID, TRY: $count"
	sleep 2
	kill -9 -$PID
done
```
After 262 successful runs, nfcpy failed as describe above on the 263th run, and then failed for all subsequent runs.

We tried the same thing but this time we used `sleep 20` and it failed on the 21th run, and then failed for all subsequent runs.

Maybe  time has some kind of influence on this behavior?

malikolivier, 2016-12-06:
As ugly as it may seem, we can still use the NFC reader and read data from it even when the above issue occurs, as this patch shows:
https://github.com/nfcpy/nfcpy/compare/master...malikolivier:dirty-fix?expand=1

nehpetsde, 2016-12-07:
The log was quite helpful as it shows that the reader itself is still responsive but returns the response for an older command. This would not happen if nfcpy terminates gracefully but may easily happen if unfriendly killed. I'd guess you're using the kill methodology as part of your application logic. That you only see it happen sparsely is most likely due to the speed of the Edison, on my workstation PC is's more or less immediate.

Fortunately the chip logic is stable enough to allow to fix this by removing any leftover response data during initialization. With your the shell script kill loop it's now running since two hours.

commit 2ecc859d (2016-12-07): Increased robustness of rcs380 to be usable after a process kill, fixes #54
closed 2016-12-07 by nehpetsde