nfcpy/nfcpy #55: Crash when init MiFare Tag

patrickFuerst, 2016-10-18:
When activating a MiFare Tag it crashes and I get the following Traceback:
```
INFO:nfc.clf:searching for reader on path usb:04cc:2533
INFO:nfc.clf:using SensorID StickID PN533v2.7 at usb:001:035
DEBUG:nfc.tag:trying to activate 106A sdd_res=0496ED52E03F80 sel_res=00 sens_res=4400
DEBUG:nfc.tag:trying type 2 tag activation for 106A
DEBUG:nfc.tag.tt2_nxp:check if authenticate command is available
DEBUG:nfc.tag.tt2_nxp:check if version command is available
Traceback (most recent call last):
  File "/home/nfc/tests/NFCWrite.py", line 37, in <module>
    tag = nfc.tag.activate(clf, tag_target)
  File "build/bdist.linux-x86_64/egg/nfc/tag/__init__.py", line 402, in activate
  File "build/bdist.linux-x86_64/egg/nfc/tag/__init__.py", line 420, in activate_tt2
  File "build/bdist.linux-x86_64/egg/nfc/tag/tt2.py", line 664, in activate
  File "build/bdist.linux-x86_64/egg/nfc/tag/tt2_nxp.py", line 689, in activate
  File "build/bdist.linux-x86_64/egg/nfc/tag/tt2_nxp.py", line 590, in __init__
TypeError: __init__() takes exactly 5 arguments (3 given)
[Finished in 2.9s with exit code 1]
```

Looks like some arguments are missing: https://github.com/nfcpy/nfcpy/blob/stable/0.11/nfc/tag/tt2_nxp.py#L590

nehpetsde, 2016-10-20:
Looks like this is a Mifare Plus card with the NXP proprietary crypto. Thiose tags can not be used with nfcpy (although it should not just fail with an exception).

mascii, 2016-10-22:
I just want to read the card ID, but same problem happened in OS X and Raspbian.

```
$ python tagtool.py
[nfc.clf] searching for reader on path usb
[nfc.clf] using SONY RC-S380/P NFC Port-100 v1.11 at usb:020:019
** waiting for a tag **
Traceback (most recent call last):
  File "tagtool.py", line 581, in <module>
    TagTool().run()
  File "/Users/USERNAME/nfcpy/examples/cli.py", line 461, in run
    while self.run_once() and self.options.loop:
  File "/Users/USERNAME/nfcpy/examples/cli.py", line 456, in run_once
    return clf.connect(**kwargs)
  File "/Users/USERNAME/nfcpy/nfc/clf/__init__.py", line 561, in connect
    result = self._rdwr_connect(rdwr_options, terminate)
  File "/Users/USERNAME/nfcpy/nfc/clf/__init__.py", line 586, in _rdwr_connect
    tag = nfc.tag.activate(self, target)
  File "/Users/USERNAME/nfcpy/nfc/tag/__init__.py", line 402, in activate
    return activate_tt2(clf, target)
  File "/Users/USERNAME/nfcpy/nfc/tag/__init__.py", line 420, in activate_tt2
    return nfc.tag.tt2.activate(clf, target)
  File "/Users/USERNAME/nfcpy/nfc/tag/tt2.py", line 664, in activate
    tag = nfc.tag.tt2_nxp.activate(clf, target)
  File "/Users/USERNAME/nfcpy/nfc/tag/tt2_nxp.py", line 689, in activate
    return VERSION_MAP[rsp](clf, target)
  File "/Users/USERNAME/nfcpy/nfc/tag/tt2_nxp.py", line 590, in __init__
    super(MF0UL11, self).__init__(clf, target)
TypeError: __init__() takes exactly 5 arguments (3 given)
```

I cloned `stable/0.10` branch instead of `master`, and this problem did not happen when I touched the same card.

```
$ python tagtool.py
[nfc.clf] searching for reader on path usb
[nfc.clf] using SONY RC-S380/P NFC Port-100 v1.11 at usb:020:019
** waiting for a tag **
Type2Tag 'Mifare Ultralight EV1 (MF0UL11)' ID=XXXXXXXXXXXXXX
```

nehpetsde, 2016-11-10:
Looks like I've wrongly interpreted the card activation response. Turns out that for Mifare Ultralight EV1 there is an error in the number of init arguments. I'll correct that but I don't have any Ultralight EV1 tag for testing, so please let me know if it's working.

closed 2016-11-10 by nehpetsde

mascii, 2016-11-11:
It's working fine. Thanks!

```
$ python tagtool.py
[nfc.clf] searching for reader on path usb
[nfc.clf] using SONY RC-S380/P NFC Port-100 v1.11 at usb:020:003
** waiting for a tag **
Type2Tag 'Mifare Ultralight EV1 (MF0UL11)' ID=XXXXXXXXXXXXXX
```

commit 540902bf (2016-11-10): This fixes #55 - wrong number of arguments when activating a Mifare Ultralight EV1 tag.