nfcpy/nfcpy #59: undefined symbol : ECDH_OpenSSL on raspbian

rotarui, 2016-12-09:
Hi,
I made a script for reading and writting on ntag213 chips , and It worked fine on my computer. I wanted to use it on my raspberry pi but I can't get the nfc module loaded. 
Getting this error : ```AttributeError: /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.1: undefined symbol: ECDH_OpenSSL```And I can't find any answer on my specific problem :/ 
Do you have any ideas ?
Thanks !

rotarui, 2016-12-09:
Ok so I found that some symbols have been removed in newer versions of libcrypto, I used libcrypto.so.1.0.0 instead of libcrypto.so.1.1 and it worked ! Sorry for useless issue, hope it'll help someone else too !

nehpetsde, 2016-12-15:
Glad you figured out. I've checked a recent Raspbian install and it's still on libcrypto 1.0, so there should not normally be a problem. But it's good to know there's a problem ahead, hopefully it's solvable by then. Thanks for pointing it out. I've added a safety check to prevent silent use of the newer version.

closed 2016-12-15 by nehpetsde

jayjohns0n, 2018-02-26:
Hello, I have the same problem. Can you explain me how to change to libcrypto.so.1.0.0?

faddey-w, 2018-08-27:
Faced the same issue too. Not sure that I'm using the recent version of the library, but it looks like attempt to use "ECDH_OpenSSL" is still present in current master version of nfcpy, so I think it is relevant.

Some details on my system (ask me if you need something else):

`$ uname -a`
```
Linux raspberrypi 4.14.30-v7+ #1102 SMP Mon Mar 26 16:45:49 BST 2018 armv7l GNU/Linux
```

`$ cat /etc/os-release`
```PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"```

`$ python /etc/node-red/rfid/nfcpy/examples/tagtool.py`
```Traceback (most recent call last):
  File "/etc/node-red/rfid/nfcpy/examples/tagtool.py", line 37, in <module>
    from cli import CommandLineInterface
  File "/etc/node-red/rfid/nfcpy/examples/cli.py", line 35, in <module>
    import nfc
  File "/etc/node-red/rfid/nfcpy/nfc/__init__.py", line 28, in <module>
    from clf import ContactlessFrontend
  File "/etc/node-red/rfid/nfcpy/nfc/clf/__init__.py", line 34, in <module>
    import nfc.llcp
  File "/etc/node-red/rfid/nfcpy/nfc/llcp/__init__.py", line 31, in <module>
    from llc import LOGICAL_DATA_LINK, DATA_LINK_CONNECTION
  File "/etc/node-red/rfid/nfcpy/nfc/llcp/llc.py", line 41, in <module>
    from . import sec
  File "/etc/node-red/rfid/nfcpy/nfc/llcp/sec.py", line 490, in <module>
    OpenSSL = OpenSSLWrapper(libcrypto)
  File "/etc/node-red/rfid/nfcpy/nfc/llcp/sec.py", line 191, in __init__
    self.crypto.ECDH_OpenSSL.restype = c_void_p
  File "/usr/lib/python2.7/ctypes/__init__.py", line 375, in __getattr__
    func = self.__getitem__(name)
  File "/usr/lib/python2.7/ctypes/__init__.py", line 380, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.1: undefined symbol: ECDH_OpenSSL```