nfcpy/nfcpy #60: ndef.message_encoder does not have an 'error' argument

jd4ever, 2017-03-27:
Hi Stephen,

Thanks for putting together this wonderful NFC library!

I'm on the PN532 and reading works. However, when following the [example](http://nfcpy.readthedocs.io/en/latest/modules/tag.html#nfc.tag.Tag.NDEF) of writing to the tag

```
if tag.ndef is not None:
    for record in tag.ndef.records:
        print(record)
    if tag.ndef.is_writeable:
        from ndef import TextRecord
        tag.ndef.records = [TextRecord("Hello World")]
```

it returns an Exception with `TypeError: message_encoder() got an unexpected keyword argument 'errors'`

It looks like the message_encoder() of the [ndef library](https://ndeflib.readthedocs.io/en/stable/index.html) doesn't accept an argument named error:
[https://ndeflib.readthedocs.io/en/stable/ndef.html#message-decoder](url)

nehpetsde, 2017-03-27:
Thanks for raising this issue. It is already fixed in the master branch scheduled for the 0.13 release. You've been reading this example from the 0.13.dev0 documentation (latest) and that's not matching with the nfcpy version 0.12 that you've installed with pip. However, as I really want to replace the old ndef submodule with ndeflib, I should have already made a new release. This will now be coming.

nehpetsde, 2017-03-27:
Version 0.13.0 is now released and the example should be working as intended.

closed 2017-03-27 by nehpetsde

jd4ever, 2017-03-27:
Perfect. Thank you!