nfcpy/nfcpy #62: PN532 Polling slows down as inactive time increases

jerroldlaw, 2017-05-15:
I am writing an application which runs on a Raspberry Pi 3 (Jessie) with a PN532 on UART. The application polls continuously for a tag with the following settings:

rdwr_options = {
    'targets': ['106B'],
    'on-connect': lambda tag: False,
	'iterations': 1,
	'interval': 1
}

tag = clf.connect(rdwr=rdwr_options)

The application runs in a loop where the ContactlessFrontend will attempt to connect again after a tag is scanned. All is fine when tags are being scanned actively (short intervals of a few minutes at most). The problem occurs when a tag is not scanned after an hour or so, and the issue gets worst in relation to the inactive time.

After an hour or so, it takes about 5 seconds to scan a tag. After 24 hours, it takes about ~20 seconds to scan a tag, and ~10s on the subsequent scan. The time it takes to scan gets reduced as more scans are connected and will eventually work as intended in the settings. (After about 5-6 scans it will scan in the interval and iterations of 1 as inputted in the settings).

Thanks!

nehpetsde, 2017-05-16:
That's an interesting behaviour that I've not yet seen probably due to the time it takes to observe :-) Though I have a RPi music box with NFC interface (for station change) and not yet suffered from such issue. But it's only running few hours a piece.

I'll try to reproduce the issue and get back.

nehpetsde, 2017-05-22:
I could not reproduce this with my RPi music player running longer times with similar logic (connect and read a tag whenever it comes). On second thought I've noticed that you are setting iterations to 1 and interval to 1 second. But interval is only applied between iterations, hence never in your case (see _iterations_ and _interval_ documentation for [connect](http://nfcpy.readthedocs.io/en/stable/modules/clf.html#nfc.clf.ContactlessFrontend.connect)). If you don't sleep in your loop over connect then you are scanning tags as fast as RPi can go and maybe this is building up processing delay at other places. I'm not fully sure if that is the core problem but can you please experiment with either short sleep after connect or an interval greater 1 and share the results.

jerroldlaw, 2017-05-28:
Alright. I will try that and share my results. Thanks!

nehpetsde, 2017-06-12:
Any results? I'd like to close this if resolved.

jerroldlaw, 2017-06-13:
Hi,

I've tested it but I'm still getting the same problem. However, I managed to make it work using the sense function in an infinite loop directly instead of connect. I suspect it might be a hardware issue though. Thanks.

closed 2017-06-14 by nehpetsde