nfcpy/nfcpy #63: Rebooting RPi3 with clf connected to the PN532 on UART locks the serial port

dennis-ng, 2017-05-16:
While clf is open, rebooting or shutting down a raspberry pi 3 will cause the serial port (e.g. ttyAMA0) to be in a deadlock as the clf was not safely closed. This also happens when a sigkill is issued on a process using clf. Currently the only way to free up the port is to reconnect the PN532 but a reboot doesn't cut the power to the PN532.

ibanezgomez, 2017-05-16:
I have also experienced similar problems when the device and the reader have been working for a long time (> 30 days). The reader does not respond and it is necessary to remove the reader power so that the connection can be made again.

Think this may be a PN532 problem, not nfcpy. My solution has been restarting using external hardware.

Any information on this subject would be very well received, I currently have a large number of devices in operation and is a widespread failure.

jerroldlaw, 2017-05-16:
Hi ibanez. What made you think its a PN532 problem? I've tried to replicate this issue on the libnfc library but it does not seem to cause a port deadlock when killing the process / rebooting the raspberry pi. I love using the nfcpy library and would prefer to use it though.

ibanezgomez, 2017-05-16:
Hi jerroldlaw, it's easy to recreate the problem with libnfc (I guess with nfcpy it would be something similar).

Attempts to open 10 instances of libnfc (for example the nfc-list program) at the same time on 10 different shells. Put a nfc tag on top of the reader and you will see that in each insance it throws errors (logically, they are all writing at the same time on the same serial device) but they are ever able to read the tag.

After a while (10 minutes?) You will see that all instances of nfc-list throw the same error. Something like that can not connect the device.

At this point the device is "hung". The only way it works again is to remove the power and then put it back on.

Sorry if I did not explain myself well, if you want I can put a video or some images to prove it.

nehpetsde, 2017-05-16:
It's indeed a problem with PN53x that an orderly shutdown is required to get them respond next time. With serial connection it's  even more problematic because the baudrate is iinitially ncreased and then the chip  doesnt understand commands if that was not reverted. So key is to catch the relevant signals and properly deinitialize the chip - for Ctrl-C this is done but not for hard kills. For plain KILL it's even questionable because how then to get out of a weird situation. For TERM it might be possible and could be tried with a proper signal handler. But I don't know if the Linux kernel sends TERM before KILL during shutdown.

jerroldlaw, 2017-05-17:
I've solved the issue by sending GPIO signals to the RSTPDN pin in order to reinitialize the chip when the connection timeout error occurs.

ibanezgomez, 2017-05-17:
And some software solution when the device does not respond?

nehpetsde, 2017-05-20:
On numerous attempts I've never found a reliable solution to return the device into proper state when the process got killed. It sometimes works just by chance but never when the pn532 serial baudrate was configured to higher speed rate. The only reliable solution is indeed either cutting the power or to assert the reset line as @jerroldlaw figured out.

ibanezgomez, 2017-05-23:
Simply awesome guys, I have tried to "break" the NFC whit my libnfc tests and switching reset line it comes alive.

Now I have a problem, I have to solder more than 300 NFC readers XD

nehpetsde, 2017-05-23:
The RPi's gpio-poweroff device tree overlay could be used to reset a PN532 during shutdown or reboot. As the RSTPD pin has to driven low for a reset, the gpio-poweroff must be configured active_low in config.txt:

`dtoverlay=gpio-poweroff,active_low,gpiopin=26  # gpiopin 26 is also the default`

closed 2017-06-12 by nehpetsde