nfcpy/nfcpy #64: Signature Record

sherylll, 2017-05-21:
Is there any plan to implement signature record type in the future?

For a project I'm working on I want my reader to be able to verify the identity of the one sending the request. Any suggestions where to start if I want to write the class myself?

Thanks in advance,

nehpetsde, 2017-05-22:
There is certainly a wish to implement the signature record but I do not have time to do it anytime soon. If you want to do it yourself, please note that it should be added to https://github.com/nfcpy/ndeflib (the ndef submodule within nfcpy is deprecated). Extending ndeflib with private records is [documented here](http://ndeflib.readthedocs.io/en/latest/extending.html). If you wish to contribute your implementation (**which would be very much appreciated**), do also have a look at the [Contributing](http://ndeflib.readthedocs.io/en/latest/contributing.html) section.

nehpetsde, 2018-03-16:
It has been implemented in `ndeflib` with pull request nfcpy/ndeflib#4.

closed 2018-03-16 by nehpetsde