nfcpy/nfcpy #65: Support ST M24SR64 active NFC tag with I2C - communication

hassfers, 2017-05-22:
We want to connect a ST M24SR64 to a raspberry pi and use it as a NFC tag. It communicates over I2C and uses the NDEF message style to store his data. The library should be written in Python and we would like to use your nfcpy library to build the NDEF message. we got it finally working to write and read from the ST M24SR64 with your ndeftool and the ACR 122. We also have a Arduino library for i2c-transport. Everything is writen down [here](https://github.com/crossbario/iotcookbook/tree/master/device/pi/components/nfc). 
We want to ask you now how to use nfcpy in the best way to generate and read NDEF messages.

nehpetsde, 2017-05-22:
Could you help me understand whether you want to:
a) use ndeftool and nfcpy to generate/send/receive/decode NDEF over the M24SR64 NFC interface, or
b) integrate the M24SR64 chip as a device (for tag ops) within nfcpy using the I2C interface, or
c) use ndeflib to integrate NDEF generation and decoding into some other tool operating the M24SR64.

hassfers, 2017-05-23:
i think Option b) will mostly fit what we want to do.

The question is how to use your lib at the best why. 
Is it right that we just need a special device class  for M24SR24 and a I2C_transport file?

nehpetsde, 2017-05-29:
Yes, for option b) you'll need a device class implementing the "driver" interface and add an I2C transport class. From the M24SR24 documentation it appears that the chip does provide a "wired" interface to the embedded tag implementation that uses the same APDUs as an external reader over NFC. Thus the driver for nfcpy could implement the sense() functionality such that a RemoteTarget is returned with the same parameters as if a real Tag would have been found. Then nfcpy would start to send APDUs for reading and writing of the Type4ATag content.

hassfers, 2017-06-02:
did i understand right, that the M24SR64 uses the same APDU functions, as if I use a ACR122 over usb for writing on the tag. But I just write it direct to the chip. So it should be possible to feed the nfcpy driver with an instant of the M24SR64-class and it works in the same way as if I connected a ACR122 over USB and lay down a NFC chip on it? 

What does we have to implement in the driver class and what is optional, if we just want to generate NDEF messages and read or write them down on the chip?

nehpetsde, 2017-06-06:
From what I've seen in the docuemntation there is either an NFC or I2C session that uses the same APDU commands to read/write NDEF content. Thus it appears possible to drive the chip via I2C as if it would be a discovered Type 4A Tag. To implement such, the driver needs to return the basic tag discovery data from the sense() method after opening an I2C session with the chip and implement the exchange() method to send and receive raw data. With that in place, the nfcpy TT4 implementation would behave as if a tag was found and produce the APDU commands for interaction.
The whole flow of commands can be observed from using a standard reader device and reading/writing the tag via NFC. The debug log would show all the APDU exchanges that would later be directly send through I2C.

hassfers, 2017-06-16:
so first i have to implement the sense method, that your library thinks a "usual" nfc chip is found by a NFC reader, and then i have to write the i2c - transport file to fire the raw NDEF data to the chip over i2c.
Which properties must the returned object have, that it would be accepted by the nfcpy TT4.

nehpetsde, 2017-06-21:
(sorry for the answering delay)
The easiest way to find the parameters might be to use `python examples/sense.py 106A ` with a standard reader and the ST tag in proximity. This should print a line with timestamp followed by target parameters produced by `str(nfc.clf.RemoteTarget(...))`. With those parameters initialize a RemoteTarget instance to return for sense(). Don't forget to convert the hexadecimal prints with bytearray.fromhex (or manually).

closed 2018-03-16 by nehpetsde