nfcpy/nfcpy #68: Connection fails after (any kind of) errors

sherylll, 2017-06-09:
I'm having trouble letting my programs exit gracefully. Every time an error is detected in the program (or even when terminated with ctrl-c), the next run will fail due to timeout (errno 110). Then I have to reconnect the NFC shield again. This makes debugging very difficult and I would like to know how to deal with this problem.

I'm using the NFC shield from Adafruit connected via FTDI to my PC. (I also tried this with Raspberry Pi 3, but it works very poorly, maybe 1 in 10 times the Pi can find the reader, could this be due to the FTDI cable? Pi' USB port?)

Any help will be very appreciated,

Sheryl

nehpetsde, 2017-06-09:
This https://github.com/nfcpy/nfcpy/wiki/USB-TTL-serial-adapter-on-Raspberry-Pi could be the case for the bad experience on RPi.
For the other part, when you connect to a standard PC, have you made sure that Ctrl-C is captured in a `try .. except KeyboardInterrupt` clause for letting the program exit gracefully? The PN532 chipset needs to be shutdown properly on exit, otherwise it will be unresponsive in almost all cases. You could have no problems when cancelling `examples/tagtool.py` with Ctrl-C.

sherylll, 2017-06-12:
Thanks a lot! With try:... finally: clf.close() I was able to solve the problem of PN532 getting unresponsive. But with RPi I haven't found a solution, maybe I'll try another board.

nehpetsde, 2017-06-12:
For RPi I'd recommend to use a serial connection and avoid USB. On RPi 2 the serial link will run quite fast at 921.6 kbaud out of the box. On RPi 3 it runs at 115.2 kbaud with the default configuration where the fast UART is assigned to Bluetooth and the software (mini) UART assigned to the pins on the GPIO header. The UART assignment can be flipped in `/boot/config.txt` as described in https://www.raspberrypi.org/documentation/configuration/uart.md.

henrycjc, 2017-06-28:
Hi @sherylll,

The USB rail is very temperamental on the Pi - are you able to confirm if the red LED is flashing very often or is it solid? A number of issues I encountered were due to (I believe) the fluctuating power level (as in, it was the underlying USB system not nfcpy causing the issues). This could explain the 1 in 10 scenario. If you can confirm the device is showing up with `lsusb` and a reboot fixes it _every time_, it is probably not related but still worth investigating. Unfortunately my encoder did not have UART headers (only USB) so using a raw serial connection was not an option and as such I cannot comment further. 

Also a side note make sure you take note of this from the docs for `clf` for when you can access the device:

> The connect() method returns None if there were no options left after the ‘on-startup’ functions have been executed or when the ‘terminate’ function returned a true value. It returns False when terminated by any of the following exceptions: KeyboardInterrupt, IOError, UnsupportedTargetError.

sherylll, 2017-06-28:
@henrycjc 

Hi, when I used USB on the Pi the power seemed to be quite stable. But the performance was quite poor and a reboot wasn't always able to solve the problem. So in the end I switched to UART, putting 2 resistors between the shield and the Pi to convert the voltage (I'm using the shield from Adafruit, with the breakboard there's probably no such trouble...)

But connecting via USB port to the PC works fine for me; even when errors are catched, a reboot always fixes it.

closed 2017-06-28 by sherylll