nfcpy/nfcpy #69: The PN532 blocking after this

polimex, 2017-06-13:
>>> import nfc
>>> clf = nfc.ContactlessFrontend('tty:S1:pn532')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/__init__.py", line 73, in __init__
    if path and not self.open(path):
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/__init__.py", line 147, in open
    self.device = device.connect(path)
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/device.py", line 111, in connect
    device = driver.init(tty)
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/pn532.py", line 444, in init
    return Device(chipset, logger=log)
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/pn532.py", line 189, in __init__
    super(Device, self).__init__(chipset, logger)
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/pn53x.py", line 474, in __init__
    chipset_communication = self.chipset.diagnose('line')
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/pn53x.py", line 267, in diagnose
    return self.command(0x00, data, timeout=1.0) == data
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/pn53x.py", line 179, in command
    raise IOError(errno.EIO, os.strerror(errno.EIO))
IOError: [Errno 5] Input/output error
>>> clf = nfc.ContactlessFrontend('tty:S1:pn532')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/__init__.py", line 73, in __init__
    if path and not self.open(path):
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/__init__.py", line 147, in open
    self.device = device.connect(path)
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/device.py", line 111, in connect
    device = driver.init(tty)
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/pn532.py", line 435, in init
    if not transport.read(timeout=100) == set_baudrate_rsp:
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/transport.py", line 147, in read
    LEN = frame[3]
IndexError: bytearray index out of range
>>> clf = nfc.ContactlessFrontend('tty:S1:pn532')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/__init__.py", line 73, in __init__
    if path and not self.open(path):
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/__init__.py", line 147, in open
    self.device = device.connect(path)
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/device.py", line 111, in connect
    device = driver.init(tty)
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/pn532.py", line 406, in init
    if not transport.read(timeout=initial_timeout) == Chipset.ACK:
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/transport.py", line 143, in read
    raise IOError(errno.ETIMEDOUT, os.strerror(errno.ETIMEDOUT))
IOError: [Errno 110] Connection timed out
>>> clf = nfc.ContactlessFrontend('tty:S1:pn532')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/__init__.py", line 73, in __init__
    if path and not self.open(path):
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/__init__.py", line 147, in open
    self.device = device.connect(path)
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/device.py", line 111, in connect
    device = driver.init(tty)
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/pn532.py", line 406, in init
    if not transport.read(timeout=initial_timeout) == Chipset.ACK:
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/transport.py", line 143, in read
    raise IOError(errno.ETIMEDOUT, os.strerror(errno.ETIMEDOUT))
IOError: [Errno 110] Connection timed out
>>> clf = nfc.ContactlessFrontend('tty:S1:pn532')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/__init__.py", line 73, in __init__
    if path and not self.open(path):
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/__init__.py", line 147, in open
    self.device = device.connect(path)
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/device.py", line 111, in connect
    device = driver.init(tty)
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/pn532.py", line 406, in init
    if not transport.read(timeout=initial_timeout) == Chipset.ACK:
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/transport.py", line 143, in read
    raise IOError(errno.ETIMEDOUT, os.strerror(errno.ETIMEDOUT))
IOError: [Errno 110] Connection timed out
>>> quit()

nfc-list working well on /dev/ttyS1
after this tests nfc-list stop working too

Any ideas?

nehpetsde, 2017-06-14:
Can you please provide more context. Platform, code sample, anything that might help to decipher. A good idea might also be to test with one of the examples, like tagtool.py, and see if that same error occurs.

polimex, 2017-06-14:
Ubuntu 16.04 (Armbian on Orange Pi Zero)
```
root@orangepizero:~# nfc-list
nfc-list uses libnfc 1.7.1
NFC device: pn532_uart:/dev/ttyS1 opened
1 ISO14443A passive target(s) found:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04
       UID (NFCID1): 71  b3  2d  e1
      SAK (SEL_RES): 08

root@orangepizero:~/nfcpy/examples# python ./tagtool.py --device /dev/ttyS1
[nfc.clf] searching for reader on path /dev/ttyS1
[nfc.clf] no reader available on path /dev/ttyS1
[main] no contactless reader found on /dev/ttyS1
[main] no contactless reader available

root@orangepizero:~/nfcpy/examples# python ./tagtool.py --device /dev/ttyS1:pn532
[nfc.clf] searching for reader on path /dev/ttyS1:pn532
[nfc.clf] no reader available on path /dev/ttyS1:pn532
[main] no contactless reader found on /dev/ttyS1:pn532
[main] no contactless reader available

root@orangepizero:~/nfcpy/examples# nfc-list
nfc-list uses libnfc 1.7.1
NFC device: pn532_uart:/dev/ttyS1 opened
1 ISO14443A passive target(s) found:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04
       UID (NFCID1): 71  b3  2d  e1
      SAK (SEL_RES): 08

```

nehpetsde, 2017-06-14:
The syntax for --device is `--device tty:S1:pn532`. Then also run `python ./tagtool.py --device /dev/ttyS1:pn532 -d nfc.clf -v` and post the debug log.

polimex, 2017-06-14:
Here is the requested info:
```
root@orangepizero:~/nfcpy/examples# python ./tagtool.py --device /dev/ttyS1:pn532 -d nfc.clf -v
[main] enable debug output for 'nfc.clf'
[nfc.clf] searching for reader on path /dev/ttyS1:pn532
[nfc.clf] no reader available on path /dev/ttyS1:pn532
[main] no contactless reader found on /dev/ttyS1:pn532
[main] no contactless reader available
root@orangepizero:~/nfcpy/examples# nfc-list
nfc-list uses libnfc 1.7.1
NFC device: pn532_uart:/dev/ttyS1 opened
1 ISO14443A passive target(s) found:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04
       UID (NFCID1): 71  b3  2d  e1
      SAK (SEL_RES): 08
```
No any difference.

nehpetsde, 2017-06-14:
Pls replace `--device /dev/ttyS1:pn532` with `--device tty:S1:pn532`.

polimex, 2017-06-14:
Hi,
We have some progress:
```
root@orangepizero:~/nfcpy/examples# python ./tagtool.py --device tty:S1:pn532 -d nfc.clf -v
[main] enable debug output for 'nfc.clf'
[nfc.clf] searching for reader on path tty:S1:pn532
[nfc.clf.transport] trying /dev/ttyS1
[nfc.clf.transport] avail: /dev/ttyS1
[nfc.clf.device] trying pn532 on /dev/ttyS1
[nfc.clf.transport] >>> 000000000000000000000000ff02fed4022a00
[nfc.clf.pn532] wait 100 ms for data on /dev/ttyS1
[nfc.clf.transport] <<< 0000ff00ff00
[nfc.clf.transport] <<< 0000ff06fad50332010607e800
[nfc.clf.transport] >>> 000000000000000000000000ff05fbd4140100001700
[nfc.clf.transport] <<< 0000ff00ff00
[nfc.clf.transport] <<< 0000ff02fed5151600
[nfc.clf.pn532] trying to set 921600 baud
[nfc.clf.transport] >>> 000000000000000000000000ff03fdd410071500
[nfc.clf.transport] <<< 0000ff00ff00
[nfc.clf.transport] <<< 0000ff02fed5111a00
[nfc.clf.transport] >>> 0000ff00ff00
[nfc.clf.pn532] changed uart speed to 921600 baud
[nfc.clf.pn532] Diagnose 00000102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333435363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5c5d5e5f606162636465666768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838485868788898a8b8c8d8e8f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4b5b6b7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d4d5d6d7d8d9dadbdcdddedfe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2f3f4f5f6f7f8f9fafbfcfdfeff000102030405 1.000s
[nfc.clf.transport] >>> 0000ffffff0109f6d40000000102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333435363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5c5d5e5f606162636465666768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838485868788898a8b8c8d8e8f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4b5b6b7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d4d5d6d7d8d9dadbdcdddedfe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2f3f4f5f6f7f8f9fafbfcfdfeff0001020304059d00
[nfc.clf.pn532] input/output error while waiting for ack
[nfc.clf.device] [Errno 5] Input/output error
[main] no contactless reader available
root@orangepizero:~/nfcpy/examples# nfc-list
nfc-list uses libnfc 1.7.1
NFC device: pn532_uart:/dev/ttyS1 opened
1 ISO14443A passive target(s) found:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04
       UID (NFCID1): 71  b3  2d  e1
      SAK (SEL_RES): 08
```

nehpetsde, 2017-06-14:
It looks like everything worked fine until the serial speed is set up to 921.6 kbaud. It's probably working if you set `change_baudrate = True` in line 389 of [nfc/clf/pn532.py](https://github.com/nfcpy/nfcpy/blob/master/src/nfc/clf/pn532.py). If that is the case, could you then try to find out  how the OrangPi platform could be identified and fixed to 115.2 kbaud similar to what is done for Raspberry Pi 3 on lines 390-400 of pn532.py.

closed 2018-03-16 by nehpetsde