nfcpy/nfcpy #70: Not detecting PN532 using USB serial devices on macOS.

lightbord, 2017-06-24:
nfcpy isn't recognizing a PN532 connected to a usb-serial device (in my case the Adafruit FTDI friend), on macOS. It generates a device file such as `/dev/tty.usbserial-XXXXXXXX` (where X represents the serial number of the device. Specifying it with the `--device=tty:usbserial-XXXXXXXX:pn532` parameter does not work. I have already tested and confirmed that it's not the hardware, as it works fine with libnfc 1.7.1 on my Mac as well as nfcpy itself when connected to my Raspberry Pi.

nehpetsde, 2017-06-26:
Is the number XXXXXXXX generated as part of the device name a fixed length format with leading zeros or just a as many digits as needed?

lightbord, 2017-06-26:
@nehpetsde I'm not exactly sure. I know its the serial number of the device. On macOS this can be found a few ways, lots of times using the "IORegistryExplorer" developer tool from Apple). I don't have another device to test it with that would have a different length serial number, or another USB-Serial device that generates this kind of name at all). I'd think its a safe bet that at least all Adafruit FTDI Friends would have the same format. As an example mine is "AH30KIIV" (I've switched the order of some of the digits and characters around, but there are still numbers where there should be numbers and letters where there should be letters).

nehpetsde, 2017-06-27:
@lightbord Unfortunately the current implementation was assuming decimal literals at the end of the device name (like `ttyUSB0`) to realize various kinds of globbing. It'll take me a little longer to modify for non-numerical name tails and change the test cases.

nehpetsde, 2017-06-29:
Could you please test with the latest version from GitHub:
```sh
$ git clone git@github.com:nfcpy/nfcpy.git
$ cd nfcpy && python setup.py bdist_wheel
$ virtualenv testenv
$ source testenv/bin/activate
(testenv) $ pip install dist/nfcpy-0.13.1-py2-none-any.whl
```
It should then be possible to (i) autodetect FTDI serial devs `--device tty`, (ii) open a specific one `--device tty:usbserial-XXXXXXXX` or open whatever name from /dev `--device tty:link-to-my-serial-device`.

lightbord, 2017-06-30:
Success! Thanks!

nehpetsde, 2017-07-13:
I've learned from https://stackoverflow.com/questions/8632586/macos-whats-the-difference-between-dev-tty-and-dev-cu that `/dev/cu.xyz` is the version to be used for this style of access.

closed 2017-07-13 by nehpetsde