nfcpy/nfcpy #71: can't get uid from my Android mobile phone in 0.13.0.

wachaz, 2017-06-25:
I can't get uid from my Android mobile phone on 0.13.0.

It was Type3Tag on 0.9.0.
So I can get idm from the phone.

However it is Type4Tag on 0.13.0.
So I can't get idm from the phone.

tag.identifier returns not valid value.
for example, "083d775e".

and also tagtool.py doesn't display uid from 0.13.0( maybe from 0.10.0 ).
```
Ver.0.9.0
$ examples/tagtool.py
[nfc.clf] searching for reader with path 'usb'
[nfc.clf] using SONY RC-S380/P at usb:001:007
[main] touch a tag
Type3Tag IDm=XXXXXXXXXXXXXXXX PMm=XXXXXXXXXXXXXXXXX SYS=fe00
```
``` 
Ver.0.13.0
$ examples/tagtool.py 
[nfc.clf] searching for reader on path usb
[nfc.clf] using SONY RC-S380/P NFC Port-100 v1.11 at usb:001:007
** waiting for a tag **
Type4ATag MIU=255 FWT=0.03XXXX
```

nehpetsde, 2017-06-29:
Have you already tried to set `--technology F`? When polling a phone for a tag response the result generally depends aon (i) the capabilities of the phone and (ii) which technology response is returned first.

Generally it might be helpful to have look at the debug log (`-d nfc.clf`).

wachaz, 2017-07-01:
Thank you!
It is working fine now!

closed 2017-07-01 by wachaz