nfcpy/nfcpy #72: couldn't accept mifare and felica

wachaz, 2017-07-10:
I would like to make a software that accepts mifare and felica on same time.

After the issue #71 solved, I connected the device with a target "212F".
Then I can read my mobile phone as Type3Tag.
After changing, it was not able to read mifare.
It is obviously.

How ever I would like to make a software that accepts mifare and felica on same time.
I tried to make 2 threads to read felica and mifare.
2 threads mean, one connected with a target "212F", the other connected with a target "106A".
Bit it didn't work.
I connected with a target [ "106A", "212F" ], then my mobile phone was detected as Type4Tag.

Are there any way to accept mifare and felica without detecting my mobile phone as Type4Tag?

Thank you.

nehpetsde, 2017-07-10:
For NFC tag discovery the reader must generally poll (sense) sequentially for the different technologies (A, B and F). It there is a target in the RF field then a response is received after polling for that technology. If the target supports multiple technologies, as in your case with the smart phone, it is free to choose to which technology to respond, generally this will be the first supported. If the target is brought into range while the reader is polling, the result is naturally random. So strictly speaking it is impossible to have a reader simultaneously detect targets of two different technologies. Though it is possible to implement a different polling strategy where the reader polls until a first response, drops the RF field to reset all targets and then polls for the next technology. Finally the reader can chose the technology at which to activate the target or, if there a re two distinct targets in the field, the one to activate. This kind of loop you'd have to implement directly with the clf.sense methods, something to start with might be the low level polling code in examples/sense.py.

wachaz, 2017-07-11:
I am trying it.

All article in Japanese that I read said we should use nfcpy 0.9.0.
I felt it is not a good solution.

Thank you.

closed 2017-07-11 by wachaz