nfcpy/nfcpy #73: Importing termios prevents operation on Windows

svvitale, 2017-07-11:
Hi @nehpetsde,

I mostly use nfcpy on Linux, but sometimes I do a little Windows development too.  I noticed that in the latest PyPI version, termios is imported.  Unfortunately it's not present or supported on Windows.  I don't know enough about how you're using it to know what's the best way to fix it.

The 0.13.0 release works fine, only 0.13.1 is affected.

nehpetsde, 2017-07-12:
Many thanks for the bug report. Release 0.13.1 has changed local device recognition for better support of Mac OS and that modified the file where termios is used. I must have shifted the import to where it's no longer dependent on operating system. Will fix soon. Thanks again, very helpful.

nehpetsde, 2017-07-12:
Release 0.13.2 is supposed to fix the issue. I could not test it on a Windows machine but was confident enough about the correction to dare the full release cadence. I'd appreciate if you could verify. Thanks again @svvitale for raising the issue.

svvitale, 2017-07-12:
Tested and verified that 0.13.2 works as expected.  Thanks for fixing it so quickly!

closed 2017-07-12 by svvitale