nfcpy/nfcpy #74: iterations and interval options in rdwr option not working.

wachaz, 2017-07-13:
iterations and interval options in rdwr option not working.

in nfc/clf/__init__.py, 
```
560:    def _rdwr_connect(self, options, terminate):
561:        target = self.sense(*options['targets'], iterations=5, interval=0.5)
```
iterations and interval in options looks ignored.

Thank you.

wachaz, 2017-07-13:
to see this problem,
In the below program, the term_func was called every two seconds, any interval and any iterations.

```
# -*- coding: utf-8 -*-
import datetime
import sys
sys.path.append( '/usr/lib/python2.7/nfc/trunc' )
import nfc

t0 = datetime.datetime.now()

def term_func():
	t = datetime.datetime.now()
	print "terminater:" + str( ( t - t0 ).total_seconds() ) + "sec"
	return False

clf = nfc.ContactlessFrontend('usb')
tag = clf.connect( rdwr={ "interval": 0.1, "iterations": 2  }, terminate = term_func )
```

Thank you.

nehpetsde, 2017-07-13:
You seem to be on a release before version 0.12.0. The config options 'iterations' and 'interval' for clf.connect() were added in version 0.12 but existed before for the low level clf.sense() method. Since version 0.12 the relevant code section is:
```python
def _rdwr_connect(self, options, terminate):
    target = self.sense(*options['targets'],
                        iterations=options['iterations'],
                        interval=options['interval'])
```

wachaz, 2017-07-13:
Sorry,
it was my mistake...

closed 2017-07-13 by wachaz