nfcpy/nfcpy #76: Use with SPI bus

falkartis, 2017-07-19:
I'm trying to use a PN532 board over the SPI bus on a raspberry pi.

I've followed this tutorial https://www.itead.cc/blog/raspberry-pi-drives-itead-pn532-nfc-module-with-libnfc And managed to get the ID from a mifare tag, but I didn't find a way to read it's content.
After a lot of searching I found this repo but it seems it always looks for a usb connected reader.

Is it possible to use this library over with an SPI or I2C connected board?

nehpetsde, 2017-07-20:
It is not currently possible to connect PN532 boards over SPI or I2C. The only option is High Speed UART (HSU) connection. When using the RPi hardware UART the serial link will run at 921.6 kbit/s.

falkartis, 2017-07-23:
Ok, and does the PN532 board support this HSU?

nehpetsde, 2017-07-23:
All boards I've seen have all three of HSU, I2C and SPI broken out. Sometimes pins are double assigned.

falkartis, 2017-07-24:
Lots of thanks.

How do I tell the library to look for HSU instead of USB?

nehpetsde, 2017-07-24:
See http://nfcpy.readthedocs.io/en/latest/overview.html#supported-devices for the path setting in case of an Adafruit PN532 board. The path goes into `ContactlessFrontend.open()` or `--device` option for the example scripts.

falkartis, 2017-07-24:
Hi, I've made a litle step of progress.
Now I have the following:

```
$ sudo python ./tagtool.py  --device "tty:AMA0:pn532" -d nfc.clf -v
[main] enable debug output for 'nfc.clf'
[nfc.clf] searching for reader on path tty:AMA0:pn532
[nfc.clf.transport] ^ttyAMA0$
[nfc.clf.transport] check: /dev/ttyAMA0
[nfc.clf.transport] avail: /dev/ttyAMA0
[nfc.clf.device] trying pn532 on /dev/ttyAMA0
[nfc.clf.pn532] running on Raspberry Pi
[nfc.clf.transport] >>> 000000000000000000000000ff02fed4022a00
[nfc.clf.pn532] wait 100 ms for data on /dev/ttyAMA0
[nfc.clf.transport] <<< 00ff00ff000000ff06fa
[nfc.clf.device] [Errno 19] No such device
[main] no contactless reader found on tty:AMA0:pn532
[main] no contactless reader available
```

It seems the chip is answering, `[nfc.clf.transport] <<< 00ff00ff000000ff06fa`, but I still can't manage to make it work. Is my chip broken? or what am I missing?

nehpetsde, 2017-07-25:
Something is screwed up. The first 6 octets to read must be a chipset acknowledge `0000ff00ff00` and it looks like the first octet is missing. All wired correctly and DIP switches set to UART? And is this reproducible?

falkartis, 2017-07-26:
I don't know, some manufacturers  swap the labels from RX and TX, so I tried both combinations, one shows what I posted, and the other shows even less information. Anyway, now I'm trying to work with a 522 chip and I feel more luck.

LeSanchez, 2019-06-03:
Hey @nehpetsde I was wondering if the SPI interface support had been added to this library? Adafruit's site is strongly [recommending the use of the SPI bus](https://learn.adafruit.com/adafruit-pn532-rfid-nfc/python-circuitpython#circuitpython-installation-of-pn532-library-5-18) for this reader.

nehpetsde, 2019-06-04:
It has not been added. And I can not confirm any serious issues using PN532 with RPi over UART. I've no ide why Adafruit is making such strong statement but it could be specific issues with their PN532 library (which looks tailored to CircuitPython).

santtu, 2020-03-25:
Just wanted to note that the problems @falkartis may have had with UART can be due to the UART0 on some Raspberry Pis being tied to the bluetooth chip (see https://www.raspberrypi.org/documentation/configuration/uart.md).

For example, on a Raspberry Pi the board needs to be connected via the mini-UART port. `tty:serial0:pn532` or `tty:ttyS0:pn532`. 

Just FYI in case someone stumbles on this issue later.