nfcpy/nfcpy #77: Do you have any plan to support PN7150?

jctlmm, 2017-07-19:
Hi,

We are planning to use the PN7150 part in the next revision of our product and we are wondering if it is in your plans to support it? FYI, on our current revision we are using PN532...

Thanks,

-Juan

nehpetsde, 2017-07-20:
I would love to have it supported but (i) the PN7150 host communication protocol is NCI which is not yet implemented, (ii) I do not have any unit available and (iii) my focus is currently to reach sufficient test coverage for reliably porting nfcpy to Python 3. Fundamentally it should not be too difficult to implement NCI but for now it's unlikely to fit my time budget. I'd of course be happy to see, review and support code contributions :-)

jctlmm, 2017-07-20:
Thanks for the information!!!

peshi, 2017-08-05:
I am also looking for this

daskalou, 2018-03-09:
Would be great to support this as NXP recommend using PN7150 now instead of the PN53* series.

I can have a look at NCI implementation if you point me in the right direction.

nehpetsde, 2018-03-16:
For nfcpy this would mean to implement a device driver that provides the interface of the class [`nfc.clf.device.Device`](https://github.com/nfcpy/nfcpy/blob/6ebb26b8f3a47183a66002b05a2c780a7e861b24/src/nfc/clf/device.py#L130). Because NCI will be common to a range of devices, the generic NCI stuff should be separated.

src/nfc/clf/nci.py:
```python
from . import device
class Device(device.Device):
      pass
```

src/nfc/clf/pn7150.py:
```python
from . import nci
class Device(nci.Device):
    pass
```

Then of course this requires knowledge of the NCI Specification and NXP's documentation and some reverse engineering from existing NCI implementations (Linux kernel, Android, ?elsewhere?) could also be helpful.

seanlano, 2018-11-19:
For anyone who is interested, NXP have an example project using the PN7150 over USB: https://www.nxp.com/documents/software/SW4335.zip

I'm only very new to this kind of NFC stuff, but if I can learn enough about how NCI works maybe I can one day help out with a driver for nfcpy.

nehpetsde, 2018-11-29:
That would be a great contribution. The device driver interface is defined by the Device class in `src/nfc/clf/device.py`.

milutz, 2019-03-03:
I'm also after PN7150 support and just at the first stages at wrapping my head around `nfcpy`, but I did take a look at the zip that @seanlano pointed at - that looks likes it might be some _good stuff_!

After unzipping my way down (NXP really seems to love zips in zips) there is source examples for window, macOS and Linux. Digging down the linux source, the file `NXP-NCI_Linux_example/NfcLibrary/NxpNci/src/NxpNci.c` looks promising as NCI guide.

Logixor, 2019-04-01:
Also working with the PN7150 here, struggling to make it work with Python.

For me at this point the best resolution is something similiar to this first integration with Python:
https://gist.github.com/doronhorwitz/fc5c4234a9db9ed87c53213d79e63b6c

It just wraps around the nfcDemoApp using the "subprocess" module.

I came up on this from the following discussion on the linux_libnfc-nci repo itself here:
https://github.com/NXPNFCLinux/linux_libnfc-nci/issues/49

closed 2019-07-27 by nehpetsde