nfcpy/nfcpy #78: What diffrent  between libnfc

jordanmv1c, 2017-08-22:
Hi ,about this project.
1-what is the diffrent between this project and libnfc?
2-if i want to use on windws,what should i install to use that project?
3-witch type of tag can i read with that projedt? I want to read via acr122u. Can i read iso 14443 type  a or b
Thank

nehpetsde, 2017-08-23:
1 - Python vs. C
2 - See https://nfcpy.readthedocs.io/en/latest/topics/get-started.html#installation
3 - See https://nfcpy.readthedocs.io/en/latest/overview.html

closed 2017-08-23 by nehpetsde

jordanmv1c, 2017-08-23:
@nehpetsde thank about your reply
I saw those links and i see this project support on acr122u , but i dont see there witch type of iso that support?
Is that project support iso14443 type a? Iso144443 type b?

About the folder of examples and test:
I saw there files of client&server?why do i need client and server?
I looking(and not find) only simple code to read data and uid from card

Thank you friend!

nehpetsde, 2017-08-23:
Clients and servers relate to NFC peer-to-peer mode. You are looking for card reading which is performed in reader-writer mode. An exhaustive tag reader example is tagtool.py. This will read/write NDEF data with all four tag types. Tag type 4 is the mapping of NDEF on ISO14443 cards.

jordanmv1c, 2017-08-25:
@nehpetsde  thank you about your fast answer
if you can please answer me on the next questions:
 i saw this page https://nfcpy.readthedocs.io/en/latest/overview.html
1- what is the type`  1 | 2 | 3 | 4A | 4B` that wrote there please? 
2- why ACR122U is not good for nfcpy? (i want only read/write tag, not peer to peer) 
3-about the install, i dont want to use `Zadig`, all i need for install the libusb is move the dll i download from http://libusb.info/ to `C:\Windows\System32` and `C:\Windows\SysWOW64`? 
and then `pip install -U nfcpy` ? that all the install on windows?

thank you a lot!

reader are support on nfcpy , but i don't understand

nehpetsde, 2017-08-25:
1. Four tag types defined by NFC Forum. Type 1 is based on Topaz. Type 2 is
based on Mifare. Type 3 is based on FeliCa. Type 4A and 4B are based on ISO
14443.

2. ACR 122 is ok for tag reading and writing.

3. Installation is described on getting started doc.