nfcpy/nfcpy #79: Do you plan to update the code in order to support the new versions of bluez?

johnplagis, 2017-08-24:
Hello, 
first of all thank you for your efforts and this excellent library. I want to ask if you plan any updates to the code in order to support the new versions of the bluez, because the bluez-simple-agent is no more available. What I am trying to accomplish is a handover connection using a bluetooth carrier over llcp (between my system and an Android smart phone).

nehpetsde, 2017-08-25:
Unfortunately this is not easily done as the interface has pretty much changed between bluez 4 and 5. and presently I'm not having sufficient time to dive into details. Of course I'd appreciate any help.

johnplagis, 2017-08-26:
Ok If I manage to get anything working I will submit it here.
Is there any way to pass raw data to LLCP link between two peers? Using the generator from [this article](http://www.ti.com/general/docs/litabsmultiplefilelist.tsp?literatureNumber=sloa187a) I am able to generate the code I need in hex format, in order to make the handover request with a bluetooth carrier. Although I didn't manage to use it with nfcpy.

nehpetsde, 2017-08-26:
The NFC part of handover should be just working; it's the final Bluetooth connection setup that is not going to work bluez 5. If you comment the lines 40, 41, 46, 47 in examples/handover-test-client.py and call it with `python examples/handover-test-client.py --skip-local -t 04` then you should see the handover select response before it breaks in line 410 (where a BluetoothAdapter object is created).

johnplagis, 2017-09-05:
So, far I wasn't able to have any results. All I found is that nfc has completely moved out of bluez to dbus. What I am thinking is downgrading my bluez version in order to get it working. Do you remember which version of bluez was compatible? Any version below 5?

nehpetsde, 2017-09-06:
It worked fine with blue 4. I haven't tried with any earlier version.

closed 2017-09-17 by johnplagis