nfcpy/nfcpy #80: error on examples\tagtool.py

userDevop123, 2017-08-25:
i tried to run examples\tagtool.py on windows 10 64 bit

i get this error

```
C:\Users\X\Downloads\nfcpy-master\nfcpy-master\examples>python tagtool.py
Traceback (most recent call last):
  File "tagtool.py", line 35, in <module>
    from cli import CommandLineInterface
  File "C:\Users\X\Downloads\nfcpy-master\nfcpy-master\examples\cli.py", line 32, in <module>
    import nfc
  File "c:\Python27\lib\site-packages\nfc\__init__.py", line 22, in <module>
    from . import clf                                                  # noqa: F401
  File "c:\Python27\lib\site-packages\nfc\clf\__init__.py", line 25, in <module>
    from . import device
  File "c:\Python27\lib\site-packages\nfc\clf\device.py", line 32, in <module>
    from . import transport
  File "c:\Python27\lib\site-packages\nfc\clf\transport.py", line 31, in <module>
    import usb1 as libusb
  File "c:\Python27\lib\site-packages\usb1\__init__.py", line 61, in <module>
    from . import libusb1
  File "c:\Python27\lib\site-packages\usb1\libusb1.py", line 199, in <module>
    libusb = _loadLibrary()
  File "c:\Python27\lib\site-packages\usb1\libusb1.py", line 173, in _loadLibrary
    return dll_loader('libusb-1.0' + suffix, **loader_kw)
  File "c:\Python27\lib\ctypes\__init__.py", line 362, in __init__
    self._handle = _dlopen(self._name, mode)
WindowsError: [Error 193] %1 is not a valid Win32 application
```

nehpetsde, 2017-08-25:
Have you installed the libusb library as described in getting started?

userDevop123, 2017-08-26:
@nehpetsde i actually not sure if i understand the installation instructions
if i understand well if need :
1-install WinUSB with  Zadig
2- copy 2 dll from  libusb i download from  http://libusb.info/ for 2 folder on windows
3- install nfcpy with `$ pip install -U nfcpy`

do i understand well?
about step number 1:
what does it mean ( from  installation instructions)

>  select your device and choose the WinUSB driver to install

do i need to connect the reader and select it there , and then choose WinUSB?
do i need to do it per type of reader?

closed 2017-08-27 by userDevop123