nfcpy/nfcpy #81: Type4Tag with SCL3711 not working

Crazy1Floor2, 2017-08-27:
I run` tagtool.py`
when i put on SCL3711 Type4Tag i still see ` ** waiting for a tag **`
but when i put Type2Tag I see the name of card +identifier (all is good)

why i don't Type4Tag  not working for me?

nehpetsde, 2017-08-28:
I had only one type of Type4B card for test, so there may well be problems with some others. Please specify the cards you're using and also run tagtool with debug option and post output.

userDevop123, 2017-08-28:
@nehpetsde
witch type of card do you try and witch reader do you use?
if i run sense.py with 106A i see None.

i run tagtool with debug and output to file but the file is empty because it stack on waiting for card.

nehpetsde, 2017-09-06:
I've tested with a Texas Instruments chip that implements a so called dynamic tag with Type4B. Don't recall the name right now but can look it up tomorrow.

StephenTiedemann, 2017-09-07:
My one and only Type4B tag is the Texas Instruments RF430CL330H NFC Booster Pack. I've just tested again reading with SCL3711, RC-S380, RC-S330, and ACR122 - worked fine.

closed 2017-12-21 by nehpetsde