nfcpy/nfcpy #83: nfcpy with android?

userDevop123, 2017-10-10:
can I run npcpy with android? how please?
with witch reader nfcpy work?

nehpetsde, 2017-10-11:
You can run nfcpy on a Linux/Windows/Mac computer and exchange data over NFC with any Android phone. But you can not run nfcpy on an Android phone. Theoretically one could manage to run Python on Android and install nfcpy into the Python environment this would still be no use because the NFC chipsets on Android phones are not the same as in consumer reader devices.

userDevop123, 2017-10-12:
@nehpetsde all the Android phones has the same NFC chipsets ? (witch kind?)
with witch kind of NFC chipsets nfcpy works?  and do you know another project that work on android?

andychlee, 2017-10-16:
@userDevop123 [This](https://en.wikipedia.org/wiki/List_of_NFC-enabled_mobile_devices) lists android phones with their NFC controllers. And for using NFC on Android, refer to https://developer.android.com/guide/topics/connectivity/nfc/index.html

closed 2017-11-07 by nehpetsde