nfcpy/nfcpy #84: IndexError while encoding to tag

henrycjc, 2017-11-02:
Hi Stephen,

I'll preface this by saying I'm pretty sure this issue is to do with a tag losing connection with the encoder, or some other kind of interference, but its the error handling that I think needs some work.

Error as below:
```
Traceback (most recent call last):
  File "tool.py", line 42, in <module>
    ret = clf.connect(rdwr=config)
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/__init__.py", line 580, in connect
    result = self._rdwr_connect(rdwr_options, terminate)
  File "/usr/local/lib/python2.7/dist-packages/nfc/clf/__init__.py", line 611, in _rdwr_connect
    if options['on-connect'](tag):
  File "tool.py", line 19, in on_connect
    tag.ndef.records = [rec]
  File "/usr/local/lib/python2.7/dist-packages/nfc/tag/__init__.py", line 207, in records
    self.octets = b''.join(message_encoder(value))
  File "/usr/local/lib/python2.7/dist-packages/nfc/tag/__init__.py", line 234, in octets
    self._write_ndef_data(data)
  File "/usr/local/lib/python2.7/dist-packages/nfc/tag/tt2.py", line 250, in _write_ndef_data
    tag_memory[offset+index] = octet
  File "/usr/local/lib/python2.7/dist-packages/nfc/tag/tt2.py", line 641, in __setitem__
    self.__getitem__(key)
  File "/usr/local/lib/python2.7/dist-packages/nfc/tag/tt2.py", line 638, in __getitem__
    return self._data_in_cache[key]
IndexError: bytearray index out of range
```

This would occur occasionally when encoding a large number of tags. I presume it was when the device lost connection, but this should possibly be handled better. I had another exception but I am struggling to recreate the issue now. I will update this issue when I can successfully recreate it. 

Here is a slightly modified version of the script I was using: https://gist.github.com/henrycjc/c1632b2d1f210ae0ff33d860c7c2eb8f

Encoder: "ACR 122 (ACR122U-A2)"
Hardware: Raspberry Pi 3 Model B V1.2

I am using `libusb`. 

Platform: 
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux

pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
```

Lib version: 
```
pi@raspberrypi:~$ pip freeze
# ommited
ndeflib==0.2.0
nfcpy==0.13.2
```

Python version:
```
pi@raspberrypi:~ $ python --version
Python 2.7.9
```

henrycjc, 2017-11-03:
Apologies for not providing more details, I'm hoping you'll recognise the issue instantly. I'm guessing it's related to me using a "not recommended" encoder.

nehpetsde, 2017-11-07:
Henry, thanks for reporting this issue. I agree that the exception thrown is not as it should be. I still have to think about it but a likely outcome is to throw a more logical exception like `NdefWriteError`.

```python
try:
    tag.ndef.records = [rec]
except nfc.tag.NdefWriteError:
    print("failure")
```

I would still be intersted to learn more about your scenario. Are all your tags exactly the same product? Can you tell me the tag product for which the error happened? Are communication characteristic predictable or not, i.e. is it a machinery that moves tags in and out of communication range or a human? Even ACR122 should be reliable for tag operations, though I've never tested with such many tags as 2000+ in a single session. If at all about the hardware setup I would be more concerned about the RPi's USB implementation that I've sometimes found to be less reliable than usual PC hardware.

nehpetsde, 2017-11-09:
I've release version 0.13.4 which will raise nfc.tag.TagCommandError when NDEF data could not be written to the tag. Application code must capture this exception to robustly handle the error condition.
```python
from ndef import TextRecord
try:
    tag.ndef.records = [TextRecord("some data")]
except nfc.tag.TagCommandError as err:
    print("NDEF write error: {}".format(error))
```

closed 2017-11-09 by nehpetsde