nfcpy/nfcpy #85: Error with Raspberry Pi 3

aeleuterio, 2017-11-05:
Hi folks,
I have a PN532 connected to my Pi's USB port through a TTL adapter. I have followed every tutorial regarding enabling UART and setting it up. When running NFCPy's example, I get this:

` $python -m nfc --search-tty
No handlers could be found for logger "nfc.llcp.sec"
This is the 0.13.3 version of nfcpy run in Python 2.7.13
on Linux-4.9.41-v7+-armv7l-with-debian-9.1
I'm now searching your system for contactless devices
Sorry, but I couldn't find any contactless device
`

I don't even know how to start debugging the problem. Any ideas will be much appreciated.

Thank you

aeleuterio, 2017-11-05:
Turns out the problem was a bad wire..
I got it to work with Python 2.x but on Python 3 I get the following import error:

` python3 -m nfc

OpenSSL libcrypto.so.1.1 is not supported
Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.5/runpy.py", line 142, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/usr/lib/python3.5/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/home/pi/.local/lib/python3.5/site-packages/nfc/__init__.py", line 25, in <module>
    from . import snep                                                 # noqa: F401
  File "/home/pi/.local/lib/python3.5/site-packages/nfc/snep/__init__.py", line 27, in <module>
    from nfc.snep.server import SnepServer                             # noqa: F401
  File "/home/pi/.local/lib/python3.5/site-packages/nfc/snep/server.py", line 26, in <module>
    import nfc.ndef
  File "/home/pi/.local/lib/python3.5/site-packages/nfc/ndef/__init__.py", line 32, in <module>
    from nfc.ndef.record import Record
  File "/home/pi/.local/lib/python3.5/site-packages/nfc/ndef/record.py", line 36, in <module>
    from error import LengthError, FormatError
ImportError: No module named 'error'
`

nehpetsde, 2017-11-07:
Glad you've found the bad wire. As for `python3 -m nfc`, this can't work because `nfcpy` is not yet released with Python 3 support.

closed 2017-11-07 by nehpetsde