nfcpy/nfcpy #87: Trying to get run tagtool.py example

Lirux, 2017-11-28:
Hi

I followed the instructions on the "getting start" page to install everything.

I did it on a Raspberry 3 and Raspbian. Everything went fine, but when I start the script I get following error. 

```
Traceback (most recent call last):
  File "./tagtool.py", line 35, in <module>
    from cli import CommandLineInterface
ImportError: No module named cli
```

Could you please tell me which module I need to install? I googled around for a python CLI module and installed them but I still get this error.

Thanks.

nehpetsde, 2017-11-28:
The missing cli module is a file in the examples directory. Just copy to the same place as tagtool.

Lirux, 2017-11-28:
Thanks! That worked!

closed 2017-11-28 by nehpetsde