nfcpy/nfcpy #88: Import Error

Kelladros, 2017-12-10:
New to coding, so not sure if its a simple issue or not but I can't get nfc to import. I've installed everything according to the guides but just keep running into this error.

Python 2.7.14 (v2.7.14:84471935ed, Sep 16 2017, 20:19:30) [MSC v.1500 32 bit (Intel)] on win32
Type "copyright", "credits" or "license()" for more information.
>>> import nfc

Traceback (most recent call last):
  File "<pyshell#0>", line 1, in <module>
    import nfc
  File "C:\Users\skiin_000\AppData\Roaming\Python\Python27\site-packages\nfc\__init__.py", line 22, in <module>
    from . import clf                                                  # noqa: F401
  File "C:\Users\skiin_000\AppData\Roaming\Python\Python27\site-packages\nfc\clf\__init__.py", line 22, in <module>
    import nfc.tag
  File "C:\Users\skiin_000\AppData\Roaming\Python\Python27\site-packages\nfc\tag\__init__.py", line 24, in <module>
    from ndef import message_decoder, message_encoder
ImportError: cannot import name message_decoder
>>>

aguariglia, 2019-01-14:
Were you ever able to figure this out? I'm having the same issue and can't seem to figure it out

yslanive, 2019-06-13:
Hope somebody can help us because I got the same error recently and before my programs and more importantly "tagtool.py" also was working properly and all of the sudden this happened.
Grateful to receive any solution or a link were the case got reopened.

nehpetsde, 2019-06-13:
You need [ndeflib](https://github.com/nfcpy/ndeflib) - which is a dependency of *nfcpy* and normally installed automatically when you do `pip install nfcpy`. If for whatever reason that did not happen try `pip install ndeflib` or download and add to your PYTHONPATH.

Esamanoaz, 2020-07-07:
I am having this issue but I have already tried `pip install --upgrade ndeflib`

Console logs:
Traceback (most recent call last):
  File "C:\Users\esama\AppData\Local\Programs\Python\Python38-32\lib\runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\esama\AppData\Local\Programs\Python\Python38-32\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\esama\Desktop\nfc\cast_handler.py", line 16, in <module>
    import nfc_handler
  File "C:\Users\esama\Desktop\nfc\nfc_handler.py", line 4, in <module>
    import nfc
  File "C:\Users\esama\AppData\Local\Programs\Python\Python38-32\lib\site-packages\nfc\__init__.py", line 22, in <module>
    from . import clf                                                  # noqa: F401
  File "C:\Users\esama\AppData\Local\Programs\Python\Python38-32\lib\site-packages\nfc\clf\__init__.py", line 22, in <module>
    import nfc.tag
  File "C:\Users\esama\AppData\Local\Programs\Python\Python38-32\lib\site-packages\nfc\tag\__init__.py", line 24, in <module>
    from ndef import message_decoder, message_encoder
ImportError: cannot import name 'message_decoder' from 'ndef' (C:\Users\esama\AppData\Local\Programs\Python\Python38-32\lib\site-packages\ndef\__init__.py)

nehpetsde, 2020-07-08:
Have you already looked into `C:\Users\esama\AppData\Local\Programs\Python\Python38-32\lib\site-packages\ndef\__init__.py`? It should be identical to https://github.com/nfcpy/ndeflib/blob/master/src/ndef/__init__.py where you find `message_decoder` on line 103.

Esamanoaz, 2020-07-08:
I fixed this error by changing line 24 on C:\Users\esama\AppData\Local\Programs\Python\Python38-32\lib\site-packages\nfc\tag\__init__.py (nfc\tag\__init__.py)

`from ndef.message import message_decoder, message_encoder`
`#from ndef import message_decoder, message_encoder  <- original`

Looking into ndef\__init__.py does show me they are different from the github repo, so I am going to try that and revert my fix.

That also fixed my error, and will probably prevent future errors. Thanks!

Now I am also getting an error from importing nfc.
`FileNotFoundError: Could not find module 'libusb-1.0.dll'. Try using the full path with constructor syntax.`
I have downloaded a folder recommended through other closed issues called `libusb-1.0.22`
I don't understand what to do with that folder though. Could you help with that too?
[Full console logs](https://pastebin.com/sUkLHbR8)

I attempted to fix this by editing the source code for libusb1 to take an input for the path of the .dll, but then I got this error when I found the correct path:
`OSError: [WinError 193] %1 is not a valid Win32 application`

I am really confused.

Esamanoaz, 2020-07-08:
I fixed it by following this issue thread: #101

BlackLight, 2020-08-17:
I'm surprised that nobody else complained about this issue so far and nobody else bothered to provide a fix. `ndef` does indeed define `message_encoder`/`message_decoder` in `__init__.py`:

```python
message_decoder = message.message_decoder
message_encoder = message.message_encoder
```

But recent versions of Python don't blindly export symbols from `__init__.py` unless they're part of `__all__` - that's the cause of the error. I have created the pull request #169 to fix this issue:

```python
try:
    from ndef import message_decoder, message_encoder
except ImportError:
    from ndef.message import message_decoder, message_encoder
```

nehpetsde, 2020-09-03:
@BlackLight Can you provide more information about when that change to the Python import semantics has been made? So far, the `__all__` list was only to declare symbols that are to be exported for a `from module import *` statement (https://docs.python.org/3/tutorial/modules.html#importing-from-a-package).

BlackLight, 2020-09-03:
I have tested it on two installations, one on Arch Linux running Python 3.8.5 and one on Raspbian running Python 3.7.3.

Output on Python 3.8.5:

```python
>>> from ndef import message_decoder, message_encoder
>>> # No errors
```

Output on Python 3.7.3:

```python
>>> from ndef import message_decoder, message_encoder
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'message_decoder' from 'ndef' (/usr/local/lib/python3.7/dist-packages/ndef/__init__.py)
>>> from ndef.message import message_decoder, message_encoder
>>> # No errors
```

This isn't the first time however that I see such import inconsistencies when importing symbols imported in a package through `from module import *` rather than through the `__all__` symbol. I'm not really sure what policy in the interpreter causes this behaviour, but apparently other users seem to have the same problem as well.

nehpetsde, 2020-09-06:
@BlackLight I tried to reproduce with Python 3.7 but it works just fine. The issue is due to installation of the `ndef`package, which is completely different from the `ndeflib` that is installed as a dependency of `nfcpy`. The situation can be fixed by:
```
pip uninstall ndef
pip install --force-reinstall ndeflib
```

closed 2020-09-06 by nehpetsde