nfcpy/nfcpy #89: ACR1252 Compatibility

Kelladros, 2017-12-11:
Trying to get my ACR1252 reader to connect with no success, can't seem to find device in python, or access it when directed to its exact path. Not sure if the device itself is not supported or if its an issue on my end from being a new programmer. Any help would be much appreciated, thanks.

StephenTiedemann, 2017-12-20:
Have you tried `python -m nfc` as described under **Verify installation** at http://nfcpy.readthedocs.io/en/latest/topics/get-started.html#installation

Kelladros, 2017-12-20:
Running Windows 10 64 bit. What exactly do I need to type in and what do I need to type it into? I've tried everything the installation guide says to do but just keep running into syntax errors or no results. Sorry for the trouble, and thanks for the help.

nehpetsde, 2017-12-21:
I should have read more careful. The ACR 1252 is not supported by nfcpy. You could test if it works with the driver for ACR122 by adding the USB VID:PID to the `usb_device_map` in `nfc/clf/device.py`.

Kelladros, 2017-12-21:
Ok, thanks for the help. It doesn't look like adding it is going to work. Going to try and find one of the readers on the compatibility list to buy online.

nehpetsde, 2018-01-02:
The Sony RC-S380 is fully supported, /S or /P doesn't matter.

closed 2018-01-21 by Kelladros

chelinho139, 2019-06-21:
Any plans on adding support for ACR1252? Given that ACR122U is discontinued by the company and now the ACR1252 is the replacement. Thanks!

nehpetsde, 2019-06-24:
Unfortunately I do not have this reader (and no longer a budget to buy).