nfcpy/nfcpy #93: No Handlers could be found for logger "nfc.llcp.sec"

ShavedCloaca, 2018-01-10:
as stated above, this error occurs when verifying the installation and running a module.

Cheers in advance,
ShavedCloaca.

WatskeBart, 2018-08-21:
Confirming the error. Most programs do seem to work fine though.

kafffee, 2018-09-28:
I got the same problem. Anybody can help?

Thanks in advance

shimondoodkin, 2018-10-15:
i got similar error i was running turnkeylinux image my was nodejs
it is a "server" install without GUI
i guess this bluez gobject thing expects a graphical environment.

somewhat moved forward but did not solve it completely 

root@nodejs ~/nfcpy/examples# ./handover-test-client.py --device tty:ttyUSB0:pn532
No handlers could be found for logger "nfc.llcp.sec"
Traceback (most recent call last):
  File "./handover-test-client.py", line 40, in <module>
    import gobject
ImportError: No module named gobject  //////////////// notice this
root@nodejs ~/nfcpy/examples#  
root@nodejs ~/nfcpy/examples# apt install python-gobject-2 python-dbus  libdbus-glib-1-2 bluetooth bluez-tools blueman bluez python-gobject
root@nodejs ~/nfcpy/examples#  reboot

but then got error 

Method “DefaultAdapter” with signature “” on interface “org.bluez.Manager” doesn't exist in raspberry pi 

https://stackoverflow.com/questions/48898476/method-defaultadapter-with-signature-on-interface-org-bluez-manager-doesn

maybe it is because it installs now bluez 5 and not 4

kafffee, 2018-10-16:
I have no plan man

LeSanchez, 2019-06-04:
The only thing I did was `pip install -U nfcpy` and then I open the python interpreter and type `import nfc` and I get this error. Did you guys figure something out? @kafffee @WatskeBart @shimondoodkin

nehpetsde, 2019-06-04:
It is caused by logging statements in the llcp/sec module when libcrypto 1.0 is missing (or the newer 1.1 is enumerated first). This will be fixed soon. For the time being - it doesn't harm.

nehpetsde, 2019-06-05:
Fixed since version 0.13.6

closed 2019-06-05 by nehpetsde