nfcpy/nfcpy #94: How to send "check" cmd with the functions defined in tt3.py

ghxbob, 2018-01-13:
Hi,

I'd like to send following  "check" cmd to tags, is there any examples showing how to use the functions defined in tt3.py? I did not find it, hope any comment.

tag.send_cmd_recv_rsp(0xF0, "\xA5", 0.1, check_status=False)

check cmd i'd like to send:
'06' + '02 FE 00 01 02 03 04 05'(IDm) + '01'(Number IUT of Services) + '0B 00'(Service Code List) + LT
'01'(Number of Blocks) + '80 00'(Block List)

nehpetsde, 2018-01-13:
Please read first the documentation section http://nfcpy.readthedocs.io/en/latest/topics/get-started.html#read-and-write-tags. In any of those examples, once you have a tag object you can use the send_cmd_recv_rsp method on that tag object. The first argument is the command code, in your case that is 0x06. The IDm is automatically inserted (it is part of the tag representation object). All other stuff goes into the data argument.

ghxbob, 2018-01-13:
in my case, is should be 
tag.send_cmd_recv_rsp(06, 02FE000102030405 01 0B00 01 8000, 2, True, True)

I will try now~~

nehpetsde, 2018-01-15:
The IDm is not part oft data. It is insrted automaically by the Tag object as the value that was received from the Tag in connect().

closed 2018-03-16 by nehpetsde