nfcpy/nfcpy #97: Is there a way to write a short TLV?

nickaknudson, 2018-02-06:
I would like to write short TLV. Just TLV NDEF start flag (`0x03`), TLV NDEF message length, TLV NDEF message, TLV NDEF end flag (`0xFE`).

Currently, when I write a blank tag it fills with some extended TLV memory information. I am using NTAG213 and the tags are blank so I think it is coming from here:

https://github.com/nfcpy/nfcpy/blob/master/src/nfc/tag/tt2_nxp.py#L302

What is the recommended way to override this?

nehpetsde, 2018-02-06:
You could directly use `tag.write(4, b'\x03\x00\xFE\x00')` instead of `tag.format()`. If you then touch the Tag again it will be valid NDEF and accept all NDEF read/write ops. Re-touch can also programmatically be done with sense() or connect().

nickaknudson, 2018-02-06:
That works!

Thanks!

closed 2018-02-06 by nickaknudson