nfcpy/nfcpy #98: help w/ NDEF records

dogebit, 2018-02-16:
I am currently having an issue with running some of the example code provided on the [getting started page](http://nfcpy.readthedocs.io/en/latest/topics/get-started.html).

The code snippet:
```
import ndef
uri, title = 'http://nfcpy.org', 'nfcpy project'
tag.ndef.records = [ndef.SmartposterRecord(uri, title)]
```
Whenever I try to run this I get this as my output:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'NoneType' object has no attribute 'records'
``` 

The card I am using is a Type 1 card and it has NDEF capabilities (at least that's what nfcpy tells me..):
```
NDEF Capabilities:
  readable  = yes
  writeable = yes
  capacity  = 462 byte
  message   = 0 byte
```

I am using a PN532 with this, and I'm sorry if there is a simple answer to this.
(I'm a noob.)

StephenTiedemann, 2018-02-17:
What is the output of:
```python
>>> tag = clf.connect(rdwr={'on-connect': lambda tag: False})
>>> print(tag)
>>> print(tag.ndef)
```

dogebit, 2018-02-18:
When I do `print(tag)` it gives me `Type1Tag 'Topaz 512' ID=XXX`. For the other one, `print(tag.ndef)`, I get `None`- which is what the guide says you should get if the NDEF data is not present.

nehpetsde, 2018-02-20:
You said the NDEF capabilities are printed by `tagtool.py`. Have you tried to write an NDEF message with `tagtool.py --load` (you can use `ndeftool` to create an NDEF file). Also the output of `tagtool.py -dump` could be helpful.

dogebit, 2018-02-21:
Turns out I was messing something up when I was trying to write to the tag. When I try to create a file using `ndeftool` and then write it using `tagtool.py`, it works. I guess that solves the problem I'm having- thanks!

closed 2018-02-21 by dogebit