nfcpy/nfcpy #100: How to send SLP_REQ, Deselect command? seems these commands are not included in current project.

ghxbob, 2018-03-10:
Dears,

How to send SLP_REQ cmd? 
I read through RCS380.py and device.py, seems these commands are not included in current project.

Any comment?

nehpetsde, 2018-03-10:
The Tag.transcribe method can send any command.

ghxbob, 2018-03-11:
1) I'm sorry but I couldn’t find any code matching 'transcribe' in nfcpy/nfcpy，do you mean tag.transceive()?
I tried and found tag.transceive() will always treat cmd as I-block APDU, however SLP_REQ is not an APDU.

Hope for your further help.

nehpetsde, 2018-03-11:
Sorry, that was supposed to be "transceive"

nehpetsde, 2018-03-11:
Indee, looks like you have to go a bit deeper. The `ContactlessFrontent.exchange` method can be reached through the `Tag.clf` attribute. It handles raw data transfer.

ghxbob, 2018-03-14:
Contactless.Frontentend.exchange method exchange data with an ACTIVATED target,however SLP_REQ cmd is used during activation? Any comment on this issue?

nehpetsde, 2018-03-16:
Here "activated target" means either sense or listen were used and returned a target. It's not including Tag activation (Contactless.Frontentend.connect would automatically do tag activation).

nehpetsde, 2018-03-16:
Maybe you could explain more what you like to achieve.

ghxbob, 2018-03-17:
thanks for your comment, i want to write some NFC Forum digital protocol test cases with nfcpy and RC-S380, is it possible?

I have found some diffculties,
1) how to send SLP_REQ to card?
2) how to send Deselect req to card?
3) how to configure different ATS for a card emulation?

ghxbob, 2018-03-25:
ContactlessFrontent.exchange

closed 2018-03-25 by ghxbob