nfcpy/nfcpy #99: Are RATS and ATS commands included in current project？

ghxbob, 2018-03-10:
I did not find RATS command in current project, do you have plan to add RATS in sense() and ATS in listen_tta()?

nehpetsde, 2018-03-10:
Set log level to DEBUG, then quite some data will be printed. If that is not enough you can also get every data transfer printed with DEBUG-1.

closed 2018-03-10 by nehpetsde

ghxbob, 2018-03-11:
I'm sorry I have used debug logging level , however there is no ATS information printed, only SENS_RES, SDD_RES, SEL_RES.

And how to pre-set ATS command?

D:\nfcpy-0.13.4\examples>python sense.py 106A --debug
203 ms [nfc.clf] searching for reader on path usb
203 ms [nfc.clf.transport] using libusb-1.0.21
206 ms [nfc.clf.transport] path matches '^(usb|)$'
3126 ms [nfc.clf.device] loading rcs956 driver for usb:054c:02e1
3399 ms [nfc.clf.rcs956] chipset is a RCS956v1.30
3433 ms [nfc.clf.rcs956] write rf settings for 106A
3463 ms [nfc.clf] using Sony RC-S330 RCS956v1.30 at usb:001:003
3464 ms [root] add to target list: 106A
3484 ms [nfc.clf] sense 106A
3504 ms [nfc.clf] found 106A sdd_res=173246F2 sel_res=28 sens_res=0400
3506 ms [nfc.clf] found 106A sdd_res=173246F2 sel_res=28 sens_res=0400
16:31:43 106A sdd_res=173246F2 sel_res=28 sens_res=0400

ghxbob, 2018-03-11:
I also tried with python sense.py 106A --verbose

D:\nfcpy-0.13.4\examples>python sense.py 106A --verbose
319 ms [nfc.clf] searching for reader on path usb
1105 ms [nfc.clf] using Sony RC-S330 RCS956v1.30 at usb:001:003
16:35:58 106A sdd_res=173246F2 sel_res=28 sens_res=0400

nehpetsde, 2018-03-11:
Try `sense.py --atr` or `tagtool.py`

ghxbob, 2018-03-24:
I find the right class in tt4.py, thanks for your help.