nfcpy/nfcpy #101: [Error 193] %1 is not a valid Win32 application

stnfc, 2018-03-14:
Hello,
Sorry for creating this issue. 
It seems I have a common python error, but I haven't been able to find a solution.

I'm starting trying to use nfcpy and when I execute python-m nfc, I have the following error:
$ python -m nfc
Traceback (most recent call last):
  File "C:\Python27\lib\runpy.py", line 163, in _run_module_as_main
    mod_name, _Error)
  File "C:\Python27\lib\runpy.py", line 111, in _get_module_details
    __import__(mod_name)  # Do not catch exceptions initializing package
  File "C:\Python27\lib\site-packages\nfc\__init__.py", line 22, in <module>
    from . import clf                                                  # noqa: F401
  File "C:\Python27\lib\site-packages\nfc\clf\__init__.py", line 25, in <module>
    from . import device
  File "C:\Python27\lib\site-packages\nfc\clf\device.py", line 32, in <module>
    from . import transport
  File "C:\Python27\lib\site-packages\nfc\clf\transport.py", line 31, in <module>
    import usb1 as libusb
  File "C:\Python27\lib\site-packages\usb1\__init__.py", line 61, in <module>
    from . import libusb1
  File "C:\Python27\lib\site-packages\usb1\libusb1.py", line 199, in <module>
    libusb = _loadLibrary()
  File "C:\Python27\lib\site-packages\usb1\libusb1.py", line 173, in _loadLibrary
    return dll_loader('libusb-1.0' + suffix, **loader_kw)
  File "C:\Python27\lib\ctypes\__init__.py", line 366, in __init__
    self._handle = _dlopen(self._name, mode)
WindowsError: [Error 193] %1 is not a valid Win32 application

I tried with 2 different python 2.7 installer version:
-> Python 2.7.14 (v2.7.14:84471935ed, Sep 16 2017, 20:25:58) [**MSC v.1500 64 bit (AMD64)**] on win32
-> Python 2.7.14 (v2.7.14:84471935ed, Sep 16 2017, 20:19:30) [**MSC v.1500 32 bit (Intel)**] on win32

Any help would be very appreciated !

thanks in advance.

ghxbob, 2018-03-16:
Do you not use windows 7 32bit OS?

stnfc, 2018-03-16:
Many thanks for your answer.
I have a 64-bit Windows version (info from Control panel / System / System type).

nehpetsde, 2018-03-16:
Have you installed libusb as desribed in [installation](http://nfcpy.readthedocs.io/en/latest/topics/get-started.html#installation)? Especially copied 32-bit and 64-bit DLLs correctly?

> The libusb library for Windows can be downloaded from libusb (Downloads -> Latest Windows Binaries) as a 7z archive. Just unpack and copy MS32\dll\libusb-1.0.dll to C:\Windows\System32 and MS64\dll\libusb-1.0.dll to the C:\Windows\SysWOW64 directory.

stnfc, 2018-03-16:
I doubled check.
The latest binaries are libusb-1.0.21
In both Windows/system32 and Windows/SysWOW64 I have now libusb-1.0.dll, together with libusb0.dll I had before.

stnfc, 2018-03-16:
some details to be sure we are weel aligned on libusb-1.0.dll:
in C:\Windows\System32 : version 1.0.21.11156, 95KB, modified in 09-Jan 17
in C:\Windows\SysWOW64 : version 1.0.21.11156, 113KB, modified in 09-Jan 17
I you want me to check something else, do not hesitate.

ghxbob, 2018-03-17:
I met similar problem, and i turn to win 7 32bit(spent half day to move everything...)

stnfc, 2018-03-19:
Thank you for the information. I cannot change my windows version.

henrycjc, 2018-04-01:
Also experiencing this - were you able to find a fix @stnfc?

Windows 10 (Version 1709) x64
Nfcpy 0.13.4
libusb 1.0.22

stnfc, 2018-04-03:
@henrycjc 
unfortunatly not. It seems that nfcpy only support 32bits version ?

stnfc, 2018-04-05:
@henrycjc 
I have got a feedback in stackoverflow.
Seems that we have to copy to C:\Windows\SysWOW64 directory MS32\dll\libusb-1.0.dll and not MS64\dll\libusb-1.0.dll as explained in the tutorial.
The issue is now solved on my side.

anilulak, 2018-05-18:
@stnfc 
Hi. I have the same issue I tried a lot of things. But the thing that I cant understand, should we copy only "libusb-1.0.dll" or the whole directory "MS32\dll\libusb-1.0.dll" ? I have exactly the same error outpu with you.

nehpetsde, 2018-05-18:
I can not say for sure that the instructions work on more or less recent Windows 10, it's been a while ago that I tested. Hopefully in two weeks I will have a W10 machine for testing.

henrycjc, 2018-05-19:
I will make a PR with updated docs

nehpetsde, 2018-05-19:
Many thanks for the PR. Does that also mean it is actually working on Windows 10?

henrycjc, 2018-05-19:
@nehpetsde Yes, tested and working on Windows 10 (1709).

nehpetsde, 2018-05-19:
@henrycjc I double the Thanks, it really saves me a lot.

closed 2018-05-20 by nehpetsde

ascerx, 2020-02-12:
yes as @stnfc said.. its sysWow64 file into system32 folder ..spent 1 hour on this.. thanks guys for your comments