nfcpy/nfcpy #102: [Errno 19] No such device

ghxbob, 2018-03-17:
Hi,

I met a strange problem, hope anyone could help.
I have replace libusb driver successfully, and I could use SONY RC-S330, everything is ok.
when i switch to my SONY RC-S380, i always met the following error:
[Errno 19] No such device

I'm sure that my S380 works fine in another windows PC.
And the diver seems work fine. where's the problem? What should I do?

C:\Users\Administrator>python
Python 2.7.14 (v2.7.14:84471935ed, Sep 16 2017, 20:19:30) [MSC v.1500 32 bit (In
tel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import nfc
>>> clf = nfc.ContactlessFrontend('usb')
>>> print clf
Sony RC-S330 on usb:001:003
>>> clf.close()
>>> import nfc
>>> clf = nfc.ContactlessFrontend('usb')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\nfc\clf\__init__.py", line 75, in __init__

    raise IOError(errno.ENODEV, os.strerror(errno.ENODEV))
IOError: [Errno 19] No such device
>>>

ghxbob, 2018-03-18:
from python -m nfc command's rsp, i found usb:0504:06c1 at usb:001:003 access is denied, i tried to uninstall some usb diver and disable and enable the USB port, however the problem still exist.

Anyone know how to solve this problem?

nehpetsde, 2018-03-26:
Have you done the usblib driver installation for both devices? And nothing of the default Sony drivers got installed?