nfcpy/nfcpy #105: ACR122u rev. 6.x support

mdavidsen, 2018-03-29:
Hi,

I just got some new ACR122u devices, but I'm unable to connect. 

The devices identify as: 
```
Mar 30 00:21:53 rpi01 kernel: [ 1364.536887] usb 1-1.4: New USB device found, idVendor=072f, idProduct=2200
Mar 30 00:21:53 rpi01 kernel: [ 1364.536913] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Mar 30 00:21:53 rpi01 kernel: [ 1364.536922] usb 1-1.4: Product: RF1258V603 PICC Interface
Mar 30 00:21:53 rpi01 kernel: [ 1364.536945] usb 1-1.4: Manufacturer: RFCARD
```
Compared to the original ACR122u i got:

```
Mar 30 00:22:37 rpi01 kernel: [ 1408.572878] usb 1-1.2: New USB device found, idVendor=072f, idProduct=2200
Mar 30 00:22:37 rpi01 kernel: [ 1408.572895] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Mar 30 00:22:37 rpi01 kernel: [ 1408.572900] usb 1-1.2: Product: ACR122U PICC Interface
Mar 30 00:22:37 rpi01 kernel: [ 1408.572907] usb 1-1.2: Manufacturer: ACS
```
Initially the actual exception was swallowed by (nfc/clf/device.py:95), but reraising and keeping the original exception, by changing from `raise error` to `raise` we got a bit further:

```
/[...]/python2.7/site-packages/nfc/clf/__init__.pyc in __init__(self, path)
     71         self.device = None
     72         self.lock = threading.Lock()
---> 73         if path and not self.open(path):
     74             raise IOError(errno.ENODEV, os.strerror(errno.ENODEV))
     75

/[...]/python2.7/site-packages/nfc/clf/__init__.pyc in open(self, path)
    145         with self.lock:
    146             log.info("searching for reader on path " + path)
--> 147             self.device = device.connect(path)
    148             if self.device:
    149                 log.info("using {0}".format(self.device))

/[...]/python2.7/site-packages/nfc/clf/device.py in connect(path)
     87             driver = importlib.import_module("nfc.clf." + module)
     88             try:
---> 89                 device = driver.init(transport.USB(bus, dev))
     90             except IOError as error:
     91                 log.debug(error)

/[...]/python2.7/site-packages/nfc/clf/acr122.pyc in init(transport)
     59
     60 def init(transport):
---> 61     device = Device(Chipset(transport))
     62     device._vendor_name = transport.manufacturer_name
     63     device._device_name = transport.product_name.split()[0]

/[...]/python2.7/site-packages/nfc/clf/acr122.pyc in __init__(self, transport)
    170         # disable autodetection
    171         log.debug("Set PICC Operating Parameters")
--> 172         self.ccid_xfr_block(bytearray.fromhex("FF00517F00"))
    173
    174         # switch red/green led off/on

/[...]/python2.7/site-packages/nfc/clf/acr122.pyc in ccid_xfr_block(self, data, timeout)
    208         frame = struct.pack("<BI5B", 0x6F, len(data), 0, 0, 0, 0, 0) + data
    209         self.transport.write(bytearray(frame))
--> 210         frame = self.transport.read(int(timeout * 1000))
    211         if not frame or len(frame) < 10:
    212             log.error("insufficient data for decoding ccid response")

/[...]/python2.7/site-packages/nfc/clf/transport.pyc in read(self, timeout)
    303                 frame = self.usb_dev.bulkRead(ep_addr, 300, timeout)
    304             except libusb.USBErrorTimeout:
--> 305                 raise IOError(errno.ETIMEDOUT, os.strerror(errno.ETIMEDOUT))
    306             except libusb.USBErrorNoDevice:
    307                 raise IOError(errno.ENODEV, os.strerror(errno.ENODEV))

```

Anyone got any idea what could be causing this? I'm not sure what happend to the spec of the devices, but it looks like they got a new hardware revision without changing the model/product id.

nehpetsde, 2018-04-03:
The current API description (for V 2.04) on the product website looks rather similar to the V 2.02 I was using for for development. Could you run `tagtool.py -d nfc.clf -v' and post the output?

mdavidsen, 2018-04-05:
Hi,

This might tell you a lot more than it tells me. :-)

```
(nfcpy) marius@marius-lap01:~/projects/nfcpy/examples$ python tagtool.py -d nfc.clf -v
No handlers could be found for logger "nfc.llcp.sec"
[main] enable debug output for 'nfc.clf'
[nfc.clf] searching for reader on path usb
[nfc.clf.transport] using libusb-1.0.21
[nfc.clf.transport] path matches '^(usb|)$'
[nfc.clf.device] loading acr122 driver for usb:072f:2200
[nfc.clf.transport] >>> 6f050000000000000000ff00480000
[nfc.clf.transport] <<< 800a000000000002fe0041435231323255323135
[nfc.clf.acr122] initialize ACR122U215
[nfc.clf.acr122] CCID ICC-POWER-ON
[nfc.clf.transport] >>> 62000000000000000000
[nfc.clf.transport] <<< 800200000000000281003b00
[nfc.clf.acr122] Set PICC Operating Parameters
[nfc.clf.transport] >>> 6f050000000000000000ff00517f00
[nfc.clf.transport] <<< 8001000000000002fe007f
[nfc.clf.acr122] Configure Buzzer and LED
[nfc.clf.transport] >>> 6f090000000000000000ff00400e0400000000
[nfc.clf.transport] <<< 8002000000000002fe009002
[nfc.clf.acr122] Diagnose 00000102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333435363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5c5d5e5f606162636465666768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838485868788898a8b8c8d8e8f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4b5b6b7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d4d5d6d7d8d9dadbdcdddedfe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2f3f4f5f6f7f8f9fa
[nfc.clf.transport] >>> 6f030100000000000000ff000000fed40000000102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333435363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5c5d5e5f606162636465666768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838485868788898a8b8c8d8e8f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4b5b6b7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d4d5d6d7d8d9dadbdcdddedfe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2f3f4f5f6f7f8f9fa
[nfc.clf.device] [Errno 110] Connection timed out
[nfc.clf] no reader available on path usb
[main] no contactless reader found on usb
[main] no contactless reader available
```

nehpetsde, 2018-04-06:
It looks like this new version does not support the full pn532 command length. Play with the host_command_frame_max_size bytes in pn532.py and let me know at which point it gets you further.

nehpetsde, 2018-04-23:
Sorry for the delay. It looks like basic access to the PN532 is possible. The ACR device architecture differs from most other readers in that there is an additional micro-controller that may either filter commands or exhibit different time-out values. The RFConfiguration command is sent with `timeout=0.1` in `src/nfc/clf/pn53x.py#L346`. I would next try to set this to a much higher value to see if that is the reason. And for the log you could activate relative time infomtation with the `--reltime` option.

If you like to send me a device for testing I can try that but won't give any promise of when that may actually happen.

gandy92, 2020-02-02:
@mdavidsen did you get any further with this?