nfcpy/nfcpy #107: Module 'Error' not found

adamuas, 2018-04-10:
[image]

I run import nfc and I get this error.

Any clues?

nehpetsde, 2018-04-12:
Please provide more context, it's difficult to figure from the image.

lodi-g, 2018-04-19:
Hello,

Running into the same error here, with a freshly installed nfcpy:

```
$ uname -a
Linux arch 4.16.2-2-ARCH #1 SMP PREEMPT Tue Apr 17 19:37:34 UTC 2018 x86_64 GNU/Linux

$ pip3 install nfcpy --user
Collecting nfcpy
  Using cached https://files.pythonhosted.org/packages/af/6b/0cc3177ed7a92652a303174880f93050ee247f4af4303e114a211a350f21/nfcpy-0.13.1.tar.gz
Requirement already satisfied: libusb1 in ./.local/lib/python3.6/site-packages (from nfcpy) (1.6.4)
Requirement already satisfied: pyserial in ./.local/lib/python3.6/site-packages (from nfcpy) (3.4)
Requirement already satisfied: ndeflib in ./.local/lib/python3.6/site-packages (from nfcpy) (0.3.2)
Requirement already satisfied: pydes in ./.local/lib/python3.6/site-packages (from nfcpy) (2.0.1)
Installing collected packages: nfcpy
  Running setup.py install for nfcpy ... done
Successfully installed nfcpy-0.13.1

$ python3
Python 3.6.5 (default, Apr 12 2018, 22:45:43) 
[GCC 7.3.1 20180312] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import nfc
OpenSSL libcrypto.so.1.1 is not supported
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/lodi_g/.local/lib/python3.6/site-packages/nfc/__init__.py", line 25, in <module>
    from . import snep                                                 # noqa: F401
  File "/home/lodi_g/.local/lib/python3.6/site-packages/nfc/snep/__init__.py", line 27, in <module>
    from nfc.snep.server import SnepServer                             # noqa: F401
  File "/home/lodi_g/.local/lib/python3.6/site-packages/nfc/snep/server.py", line 26, in <module>
    import nfc.ndef
  File "/home/lodi_g/.local/lib/python3.6/site-packages/nfc/ndef/__init__.py", line 32, in <module>
    from nfc.ndef.record import Record
  File "/home/lodi_g/.local/lib/python3.6/site-packages/nfc/ndef/record.py", line 36, in <module>
    from error import LengthError, FormatError
ModuleNotFoundError: No module named 'error'
>>> 
```

Seems to be happening as well on a Debian Docker container.

Thanks

adamuas, 2018-04-19:
I noticed that the issue went away when I used python2.7.

Guessing it’s it has something to do with python 3.6.

lodi-g, 2018-04-19:
Yup indeed, seems to be confirmed by issue #47 .

Any ETA for the Python3 version?

Thank you

nehpetsde, 2019-06-06:
The Python3 version is out.

closed 2019-06-06 by nehpetsde