nfcpy/nfcpy #110: PN532 Issues?

jonathan-ruiz, 2018-05-18:
Hi is there any known issue with the PN532 Breakout Board from adafruit?

I'm trying to use it to emulate a TAG but I just get errors when I try to record the NDEF message.

for testing I'm using this example:
http://nfcpy.readthedocs.io/en/latest/topics/get-started.html#emulate-a-card

When I try to read and write in a real tag I do it without issues.

nehpetsde, 2018-05-18:
Please cross check with tagtool.py using the --emulate flag.

jonathan-ruiz, 2018-05-21:
Thanks for your answers,
I'm trying to dump the content of a real tag using

    tagtool.py dump -o file.ndef

But I only got an empty file, I tried with a couple of cards but didn't work. 

However emulating an empty card I can see it being read by android, when I try with iPhone I just get some feedback from tooltag.py but the iphone ignores the tag, I think this is due to the emulated card not being encoded?

is there a way to pass the ndef fields in the command line to tagtool? 

Thanks

jonathan-ruiz, 2018-05-21:
Hi I manage to get another tag, and with this oen I can read the ndef content successfully through the PN532 and copy it to a file.

I guess the other 2 cards are not fully compatible with the reader?

tried tagtool emulate using the ndef content cloned from the card and worked fine for both android and iOS.

One of the cards is the one that came with the reader actually so not sure why it would be not compatible with it, in any case I'm leaving here the data I got from this card in case some one has the same issue.

tag-type: ISO 14443-3A
Technologies: NfcA, MifareClassic, Ndef
Data Format: NXP Mifare Classic
ATQA: 0x0004
SAK: 0x08

nehpetsde, 2018-05-21:
Mifare Classic is not supported, you can use only NFC Forum Type 1/2/3/4 Tags.

closed 2019-06-06 by nehpetsde