nfcpy/nfcpy #111: iOS and bema.py

jonathan-ruiz, 2018-05-18:
Hi I'm using beam.py in the examples directory to send text string as follows

beam.py --device tty:S0 send text "tex sent"

When I read it with android every thing works fine but when I use iOS it just return an error reading the tag.
Is there any known issue with iOS?

Thanks

nehpetsde, 2018-05-18:
With beam.py you are using NFC peer to peer mode which is not supported by IOS.

jonathan-ruiz, 2018-05-21:
I'm also trying to use snep.put(...) to try to emulate a card and send the data through a PN532 NFC card connected to a RPi.

It works fine on Android but is not working on iOS, is it the same issue? 

Thanks

nehpetsde, 2018-05-23:
Yes, that's the same issue.

AndySchroder, 2024-08-11:
Does `snep.put(...)` work with recent versions of android in any way? I can't seem to get it to work (but it does work on old versions of android). Seems like card/tag emulation is the only practical solution at this time?