nfcpy/nfcpy #113: Emulate Cards

jonathan-ruiz, 2018-05-21:
I'm trying to emulate a card using a PN532 on a Rasberry Pi, my final goal is to be able to read data from the PN532 using and iOS or Android device. 

I started by testing the example here 

http://nfcpy.readthedocs.io/en/latest/topics/get-started.html#emulate-a-card

But when I try to write an NDEF register using "NXP tag writer" I get an error in the mobile app "Store failed" and in the feedback from the shell I can see a "tag release" , "tag activated" loop when I'm trying to do the writing, or when I put both devices one close to each other.

my final goal is to emulate an read only card readable from android and ios, but I think that understand this example and have it working will be helpful for me to keep going.

any Idea why it's not working or maybe another example that is closer to what I'm trying to achieve? 

Thanks

nehpetsde, 2018-05-23:
Please try first with tagtool.py --emulate to see if in general it is working as expected.

jonathan-ruiz, 2018-05-24:
This is how I tested with tagtool:
I used a normal tag and write an URL record in it with a mobile app
I checked if it worked reading the tag with a mobile app
Then I dumped the content of the tag to the RaspberryPi

    ./tagtool.py --device tty:S0 dump -o ~/tag.ndef

And then I try to emulate the tag using the file I just created

    ./tagtool.py --device tty:S0 emulate ~/tag.ndef tt3

When I read the emulated tag through a mobile the app detects the tag but theres no NDEF content.

So I tried with ndeftool

    ndeftool text 'one' text 'two' print --keep > ~/two_text_records.ndef
    ./tagtool.py --device tty:S0 emulate ~/two_text_records.ndef

But I had the same result, the emulated tag is detected and some date is read but not the ndef records

This is a screenshot of the reader with the information from the emulated card using tagtool, It shows  no ndef tags
[image]

jonathan-ruiz, 2018-05-25:
If it helps you I'm also trying with nfclib examples.
    nfc-emulate-forum-tag2

and this one worked fine on android  with this output
> NFC device: pn532_uart:/dev/ttyS0 opened
> Emulating NDEF tag now, please touch it with a second NFC device
>     In: 30  00  
>     Out: 00  00  00  00  00  00  00  00  00  00  ff  ff  e1  10  06  0f  
>     In: 30  04  
>     Out: 03  21  d1  02  1c  53  70  91  01  09  54  02  65  6e  4c  69  
>     In: 30  08  
>     Out: 62  6e  66  63  51  01  0b  55  03  6c  69  62  6e  66  63  2e  
>     In: 30  0c  
>     Out: 6f  72  67  00  00  00  00  00  00  00  00  00  00  00  00  00  
>     In: 50  00  
> HALT sent
> nfc-emulate-forum-tag2: ERROR: Software caused connection abort (103)
> nfc-emulate-forum-tag2: Success
> 

and didn't work at all for iphone with this output

> NFC device: pn532_uart:/dev/ttyS0 opened
> Emulating NDEF tag now, please touch it with a second NFC device
>     In: 30  02  
>     Out: 00  00  ff  ff  e1  10  06  0f  03  21  d1  02  1c  53  70  91

nehpetsde, 2018-05-26:
Try add '-l' and '-k' options, maybe the tag is intermediately released.

jonathan-ruiz, 2018-05-29:
Thanks, 

adding "-k -l" gave me an error
      ./tagtool.py -l -k --device tty:S0 emulate ~/tag.ndef tt3
      tagtool.py: unrecognized arguments: -k show

Adding just "-l" made it work for iPhone! But I'm reading no NDEF content with android yet

    /tagtool.py -l --device tty:S0 emulate ~/ndeftool.ndef tt3

closed 2019-07-27 by nehpetsde