nfcpy/nfcpy #115: nfcpy opens Nfc-Reader only once, than [Errno 19] No such Device

gerolon, 2018-06-05:
Hello,
i use nfcpy on Windows to communicate with  my USB-nfc-device "SCL3711". The installation from nfcpy and the usb-drivers and libs seems do be sucessfully done.

My problem is now that i only can connect the target onetime:

clf=nfc.ContactlessFrontend('usb')

works!

but if i close with:
clf.close()

it seems to doesnt work.

The next time i try again i get the following error-msg:

File "C:\SVN\NFCpython\trunk\env\lib\site-packages\nfc\clf\__init__.py", line 75, in __init__
    raise IOError(errno.ENODEV, os.strerror(errno.ENODEV))
IOError: [Errno 19] No such device

Same problem is, when i start the nfc-verification two-times:
[image]

At the second try the access is denied. 

When i remove the usb-nfc-reader and plug in again. it works again one time!

Is this a USBdriver Problem?

nehpetsde, 2018-06-05:
Plesse run 'python -m nfc -vvv' and post the output of the failing attempt.

gerolon, 2018-06-06:
Hi Stephen,
thank you for answering so quickly.

I run your command twice. The second time fails again. Here are the los from both:

(env) C:\SVN\NFCpython\trunk>python -m nfc -vvv
This is the 0.13.5 version of nfcpy run in Python 2.7.15
on Windows-7-6.1.7601-SP1
I'm now searching your system for contactless devices
DEBUG:nfc.clf.transport:using libusb-1.0.22
DEBUG:nfc.clf.transport:path matches '^(usb|)$'
INFO:nfc.clf:searching for reader on path usb:001:004
DEBUG:nfc.clf.transport:using libusb-1.0.22
DEBUG:nfc.clf.transport:path matches '^usb(:[0-9]{1,3})(:[0-9]{1,3})?$'
DEBUG:nfc.clf.device:loading pn533 driver for usb:04e6:5591
Level 9:nfc.clf.transport:>>> 0000ff00ff00
Level 9:nfc.clf.pn533:Diagnose 00000102030405060708090a0b0c0d0e0f101112131415
718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333435363738393a3b3c3d
f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5c5d5e5f606162636465
768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838485868788898a8b8c8d
f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4b5
7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d4d5d6d7d8d9dadbdcdd
fe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2f3f4f5f6f7f8f9fafbfcfdfeff000102030405
000s
Level 9:nfc.clf.transport:>>> 0000ffffff0109f6d40000000102030405060708090a0b0
0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333
363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5
5e5f606162636465666768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838
868788898a8b8c8d8e8f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaaba
aeafb0b1b2b3b4b5b6b7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d
d6d7d8d9dadbdcdddedfe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2f3f4f5f6f7f8f9fafbf
feff0001020304059d00
Level 9:nfc.clf.transport:<<< 0000ff00ff00
Level 9:nfc.clf.transport:<<< 0000ffffff0109f6d50100000102030405060708090a0b0
0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333
363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5
5e5f606162636465666768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838
868788898a8b8c8d8e8f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaaba
aeafb0b1b2b3b4b5b6b7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d
d6d7d8d9dadbdcdddedfe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2f3f4f5f6f7f8f9fafbf
feff0001020304059b00
Level 9:nfc.clf.pn533:GetFirmwareVersion  0.100s
Level 9:nfc.clf.transport:>>> 0000ff02fed4022a00
Level 9:nfc.clf.transport:<<< 0000ff00ff00
Level 9:nfc.clf.transport:<<< 0000ff06fad50333020707e500
DEBUG:nfc.clf.pn533:chipset is a PN533v2.7
Level 9:nfc.clf.pn533:RFConfiguration 0102 0.100s
Level 9:nfc.clf.transport:>>> 0000ff04fcd4320102f700
Level 9:nfc.clf.transport:<<< 0000ff00ff00
Level 9:nfc.clf.transport:<<< 0000ff02fed533f800
Level 9:nfc.clf.pn533:RFConfiguration 02000b0a 0.100s
Level 9:nfc.clf.transport:>>> 0000ff06fad43202000b0ae300
Level 9:nfc.clf.transport:<<< 0000ff00ff00
Level 9:nfc.clf.transport:<<< 0000ff02fed533f800
Level 9:nfc.clf.pn533:RFConfiguration 0400 0.100s
Level 9:nfc.clf.transport:>>> 0000ff04fcd4320400f600
Level 9:nfc.clf.transport:<<< 0000ff00ff00
Level 9:nfc.clf.transport:<<< 0000ff02fed533f800
Level 9:nfc.clf.pn533:RFConfiguration 05010001 0.100s
Level 9:nfc.clf.transport:>>> 0000ff06fad43205010001f300
Level 9:nfc.clf.transport:<<< 0000ff00ff00
Level 9:nfc.clf.transport:<<< 0000ff02fed533f800
Level 9:nfc.clf.pn533:SetParameters 00 0.100s
Level 9:nfc.clf.transport:>>> 0000ff03fdd412001a00
Level 9:nfc.clf.transport:<<< 0000ff00ff00
Level 9:nfc.clf.transport:<<< 0000ff02fed5131800
Level 9:nfc.clf.pn533:ReadRegister a000 0.250s
Level 9:nfc.clf.transport:>>> 0000ff04fcd406a0008600
Level 9:nfc.clf.transport:<<< 0000ff00ff00
Level 9:nfc.clf.transport:<<< 0000ff04fcd50700012300
[... 88 lines cut ...]
Level 9:nfc.clf.transport:>>> 0000ff04fcd4320102f700
Level 9:nfc.clf.transport:<<< 0000ff00ff00
Level 9:nfc.clf.transport:<<< 0000ff02fed533f800
I'm not trying serial devices because you haven't told me
-- add the option '--search-tty' to have me looking
-- but beware that this may break other serial devs

(env) C:\SVN\NFCpython\trunk>python -m nfc -vvv
This is the 0.13.5 version of nfcpy run in Python 2.7.15
on Windows-7-6.1.7601-SP1
I'm now searching your system for contactless devices
DEBUG:nfc.clf.transport:using libusb-1.0.22
DEBUG:nfc.clf.transport:path matches '^(usb|)$'
INFO:nfc.clf:searching for reader on path usb:001:004
DEBUG:nfc.clf.transport:using libusb-1.0.22
DEBUG:nfc.clf.transport:path matches '^usb(:[0-9]{1,3})(:[0-9]{1,3})?$'
DEBUG:nfc.clf.device:loading pn533 driver for usb:04e6:5591
DEBUG:nfc.clf.device:[Errno 13] Permission denied
** found usb:04e6:5591 at usb:001:004 but access is denied
I'm not trying serial devices because you haven't told me
-- add the option '--search-tty' to have me looking
-- but beware that this may break other serial devs
Sorry, but I couldn't find any contactless device

(env) C:\SVN\NFCpython\trunk>VVV

gerolon, 2018-06-06:
I had installed the driver with Zadic:
[image]

I had some troubles with the "libusb-1.0.dll". I have copied the dlls in the 32bit and 64bit-Folders but it works only with the 32-bit Version of Python2.7. The 64-bit Python failed with nfcpy on my computer.

Now I ordered another NFC-Reader "ACR122U" to compare the behaviour.

nehpetsde, 2018-06-09:
I could observe a similar issue on W10, different error message but also preventing a second run. For whatever reason I have not tried another reader but will do on Monday.

gerolon, 2018-06-11:
Ok, thanks. When my new ACR122U arrived here, i will post the result

nehpetsde, 2018-06-11:
From my tests today it looks like the issue is specific to PN533 based readers, those contain a USB stack within the chip itself. ACS122U is built upon a PN532 chip and worked fine w.r.t. multiple open/close cycles.

gerolon, 2018-06-12:
Ok. Very good. Thank you very much for your fast support

gerolon, 2018-06-21:
I have just tried out the ACR122U. It works!!

closed 2018-06-21 by gerolon

muhamedhfayiz, 2020-08-27:
> Ok, thanks. When my new ACR122U arrived here, i will post the result

> I have just tried out the ACR122U. It works!!

How do you solve this issue? I have same problem with windows 10. I'm user ACR122U A9 Device.

leonardovepe, 2023-02-12:
> > Ok, thanks. When my new ACR122U arrived here, i will post the result
> 
> > I have just tried out the ACR122U. It works!!
> 
> How do you solve this issue? I have same problem with windows 10. I'm user ACR122U A9 Device.

Hi @muhamedhfayiz , did you solve this? Having the same issue too.