nfcpy/nfcpy #117: NXP NT3H2211 support

hotyp, 2018-07-30:
Hi,

i use in my design NT3H2211. Writing and reading works fine, but i am unable to it with password. After deeper investigation it looks like Class NTAGI2C does not have implemented functions for that? In version map in tag_nxp.py is this device present but commented. I would like to ask you, if is possible to add this device? Thank you
 
```Python
VERSION_MAP = {
    b"\x00\x04\x03\x01\x01\x00\x0B\x03": MF0UL11,
    b"\x00\x04\x03\x02\x01\x00\x0B\x03": MF0ULH11,
    b"\x00\x04\x03\x01\x01\x00\x0E\x03": MF0UL21,
    b"\x00\x04\x03\x02\x01\x00\x0E\x03": MF0ULH21,
    b"\x00\x04\x04\x01\x01\x00\x0B\x03": NTAG210,
    b"\x00\x04\x04\x01\x01\x00\x0E\x03": NTAG212,
    b"\x00\x04\x04\x02\x01\x00\x0F\x03": NTAG213,
    b"\x00\x04\x04\x02\x01\x00\x11\x03": NTAG215,
    b"\x00\x04\x04\x02\x01\x00\x13\x03": NTAG216,
    b"\x00\x04\x04\x05\x02\x01\x13\x03": NT3H1101,
    b"\x00\x04\x04\x05\x02\x01\x15\x03": NT3H1201,
    # b"\x00\x04\x04\x05\x02\x02\x13\x03": NT3H2111,
    # b"\x00\x04\x04\x05\x02\x02\x15\x03": NT3H2211,
}
```

nehpetsde, 2018-07-31:
Unfortunately I do not have an NTAG I2C Plus for testing the code. You could try to copy and adapt the code that handles the predecessor NTAG I2C, the relevant parts are just above the VERSION_MAP. If you get something to work, feel free to submit a merge request.

TINLIANG, 2018-08-10:
Hi is it possible for you to share your codes with me about writing and reading as I want to take some reference to what I am working on. Thank you, much appreciated. My email is tl.choo@styl.solutions

closed 2019-07-27 by nehpetsde